goodrobots/maverick #728: Resize expansion must have failed

aidanjones3, 2018-02-06:
Having difficulties installing other drivers for a camera to use ORB SLAM2. Realized that the resize of the root partiition must have failed. Log file shown below. Any help would be greatly appreciated.

[maverick.dev.2018-02-06-21-37.log](https://github.com/goodrobots/maverick/files/1700889/maverick.dev.2018-02-06-21-37.log)

fnoop, 2018-02-07:
Hi, could you please run ~/software/maverick/manifests/maverick-modules/maverick_hardware/acts.d/odroid.py and post the output here?

aidanjones3, 2018-02-11:
Will post the output later today, apologize for the delay.

fnoop, 2018-03-28:
Please reopen if this is still a problem, works fine on all my testing

closed 2018-03-28 by fnoop

edwinmora, 2018-05-14:
I had this same issue. i followed the install isntructions on the goodrobots site. here is my output when i run the python script:

odroid_kernel_install_flag=no
odroid_cpucores=8
odroid_kernel4x_release=no
odroid_kernel4x_dir=no
odroid_kernel3x_backups=no
odroid_kernel_current=no
odroid_swap=0 kB
odroid_memory=2043956 kB
odroid_model=Odroid XU4
odroid_sdsize=59688
odroid_present=yes
odroid_revision=0100