goodrobots/maverick #732: visiond is re-starting without error, but not sending any stream

flouwers, 2018-02-20:
On rapsberyy 3, I try to setup visiond to stream RTSP.
I tried with 2 USB Cameras :
- Logitech Cinema HD
- UVC FPV Adapter

Automatic pipeline construction seems good for both cameras, but nothing is streamed and no error is logged in maverick-visiond.log

2018-02-20 15:41:31,807 - INFO - Starting maverick-visiond
2018-02-20 15:41:31,811 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-02-20 15:41:31,814 - DEBUG - V4l2 device input: Camera 1:2
2018-02-20 15:41:31,815 - DEBUG - driver: uvcvideo
2018-02-20 15:41:31,815 - DEBUG - card: USB2.0 PC CAMERA
2018-02-20 15:41:31,816 - DEBUG - Camera control: Brightness
2018-02-20 15:41:31,816 - DEBUG - Camera control: Contrast
2018-02-20 15:41:31,817 - DEBUG - Camera control: Saturation
2018-02-20 15:41:31,817 - DEBUG - Camera control: Hue
2018-02-20 15:41:31,818 - DEBUG - Camera control: Gamma
2018-02-20 15:41:31,819 - DEBUG - Camera control: Gain
2018-02-20 15:41:31,819 - DEBUG - Camera control: Power Line Frequency
2018-02-20 15:41:31,820 - DEBUG - Camera control: Sharpness
2018-02-20 15:41:31,821 - DEBUG - Camera format: Motion-JPEG
2018-02-20 15:41:31,824 - INFO - Camera MJPEG stream available, using mjpeg stream
2018-02-20 15:41:31,826 - INFO - Creating stream object - camera:/dev/video0, stream:mjpeg, pixelformat:YUY2, encoder:h264, size:(640 x 480 / 60), output:rtsp, brightness:0
2018-02-20 15:41:31,826 - INFO - Attaching input 'v4l2': /dev/video0
2018-02-20 15:41:31,850 - INFO - Attaching stream 'mjpeg'
2018-02-20 15:41:31,918 - INFO - Raspberry hardware decoder detected, using omxmjpegdec as mjpeg decoder
2018-02-20 15:41:32,047 - INFO - Attaching encoding 'h264'
2018-02-20 15:41:32,048 - INFO - Raspberry hardware encoder detected, using omxh264enc as h264 encoder
2018-02-20 15:41:32,315 - INFO - Attaching payload 'h264'
2018-02-20 15:41:32,322 - DEBUG - h264parse element created
2018-02-20 15:41:32,342 - DEBUG - Attaching h264pay to h264parse
2018-02-20 15:41:32,346 - INFO - Attaching output 'rtsp'
2018-02-20 15:41:32,354 - INFO - RTSP stream running at rtsp://0.0.0.0:5600/video
2018-02-20 15:41:32,358 - INFO - Pipeline: "v4l2-source /dev/video0 ! capsfilter 'image/jpeg, width=(int)640, height=(int)480, framerate=(fraction)60/1' ! jpegparse ! omxmjpegdec ! queue ! videoconvert ! raspberry-h264-encode ! x264capsfilter ! h264parse ! pay0"
**2018-02-20 15:41:32,359 - INFO - Starting camera stream
2018-02-20 15:41:32,381 - INFO - Creating RTSP factory element: <GstRtsp.RTSPUrl object at 0x760d7f48 (GstRTSPUrl at 0x73f03ba0)>
2018-02-20 15:41:52,670 - INFO - Creating RTSP factory element: <GstRtsp.RTSPUrl object at 0x760d7f48 (GstRTSPUrl at 0x73f086d8)>
2018-02-20 15:41:53,834 - INFO - Starting maverick-visiond**

I tried to manually specify the parameters, but the same happens.

flouwers, 2018-02-20:
@fnoop I tried with VLC, QGroundControl, and the WebVision tool that comes with Maverick. All fails.
I don't have any working pipeline yet.
I also tried to send the stream to a file, but it stays empty.
It could be an issue with gstreamer or rapsberry implementation, but I am lost without any other debug information.

fnoop, 2018-02-20:
Ok thanks I have a similar camera here, will try it out.  I think it's probably stalling somewhere in the pipeline, they can be very delicate to construct.

flouwers, 2018-02-21:
I managed to get a pipeline working by sending Tina file, and it looks like it is hanging (without error) with *omxmjpegdec*. It also looks like other people have the same issue with that decoder.
I'm not at all familiar with gstreamer, but next step is to try with regular jpegdec and rtsp.
Would it be possible to specify the decoder in the config file?

fnoop, 2018-02-25:
https://bugzilla.gnome.org/show_bug.cgi?id=742172
Confirmed omxmjpegdec is stalling the pipeline.  Changing it to jpegdec works, but of course hammers the cpu and increases latency.

fnoop, 2018-02-25:
Very interesting:
https://info-beamer.com/blog/omx-jpeg-decoding-performance-vs-libjpeg-turbo

fnoop, 2018-02-25:
Disabled omxmjpegdec in visiond, in master.  In stable/1.1.5, change:
            if Gst.ElementFactory.find("omxmjpegdec"):
to
            if Gst.ElementFactory.find("omxmjpegdecDISABLED"):
in ~/software/maverick/manifests/maverick-modules/maverick_vision/files/maverick-visiond

flouwers, 2018-02-26:
Hi @fnoop.
Thanks for this info!
Indeed, without omxmjpegdec the pipeline don't stall anymore.
But I still have the issue of visiond re-starting itself.
I have a working pipeline :
`gst-launch-1.0 -v v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600`
But when I specify it in maverick-visiond.conf like this :
`pipeline_override = v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600`
I got no error:

> 2018-02-26 08:11:56,066 - INFO - Starting maverick-visiond
2018-02-26 08:11:56,080 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-02-26 08:11:56,080 - INFO - Manual Pipeline Construction
2018-02-26 08:11:56,081 - INFO - Creating pipeline from config: v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600
2018-02-26 08:11:57,769 - INFO - Starting maverick-visiond
2018-02-26 08:11:57,772 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-02-26 08:11:57,773 - INFO - Manual Pipeline Construction
2018-02-26 08:11:57,773 - INFO - Creating pipeline from config: v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600
2018-02-26 08:12:00,809 - INFO - Starting maverick-visiond
2018-02-26 08:12:00,813 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-02-26 08:12:00,814 - INFO - Manual Pipeline Construction
2018-02-26 08:12:00,814 - INFO - Creating pipeline from config: v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600

I can confirm the process is not running after that:
>[dev] [mav@maverick-raspberry ~]$ ps aux |grep vision
mav        981  1.1  1.4  16004 13064 ?        Ss   Feb25  11:23 python /srv/maverick/software/maverick-fcs/tornado/webvision.py
mav       5852  0.0  0.0   4308   528 pts/9    S+   08:14   0:00 grep vision
[dev] [mav@maverick-raspberry ~]$ ps aux |grep gst
mav       5854  0.0  0.0   4308   584 pts/9    S+   08:14   0:00 grep gst

fnoop, 2018-02-26:
Hey @flouwers the pipeline you're using here is not mjpeg it's raw - is that what you intended?  You should be able to not use a manual pipeline definition and with the fix I detailed above it should autodetect everything for you.
If you do want to use a manual pipeline and it's breaking for you, if it's restarting without anything in the logs you must be hitting a code bug that for some reason doesn't seem to log.  You have to start maverick-visiond outside of the normal systemd manifest:
```
maverick stop visiond
maverick-visiond
```
If it bombs out now, you should see some sort of track in the log (~/var/log/vision/maverick-visiond.log)
Also note that you're trying to do raw video @ 60fps - you typically can't do this over USB2.

flouwers, 2018-02-27:
Hey @fnoop. I noticed raw input format was also working, even though it is not specified in camera supported formats.

I also got this pipeline working, with mjpeg :
`pipeline_override = v4l2src device=/dev/video0 ! image/jpeg, width=640, height=480, framerate=60/1 ! jpegparse ! jpegdec ! queue ! videoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600`
and raw:
`pipeline_override = v4l2src device=/dev/video0 ! video/x-raw, width=640, height=480, framerate=60/1 ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=192.168.1.14 port=5600`
Notice the removed quotes for input format. 

With visiond auto pipeline contruction, log is showing that :
`0:00:00.343306472 14534  0x1a75260 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<v4l2-source> error: Internal data stream error.`
`0:00:00.343406577 14534  0x1a75260 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<v4l2-source> error: streaming stopped, reason not-negotiated (-4)`
`Error: gst-stream-error-quark: Internal data stream error. (1) gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2-source:
streaming stopped, reason not-negotiated (-4)`

BTW, 60 FPS looks OK at that resolution.

fnoop, 2018-02-27:
Surprised you can get RAW 60 fps over usb2, it must be stripping something out!
What does the maverick-visiond log give for the constructed pipeline, when you get that error?

fnoop, 2018-02-27:
I verified with (original) sj4000 camera over mjpeg and it works for me

flouwers, 2018-02-27:
How do I update to your latest change (omxmjpeg disabled)? Even after a 'maverick self-update' the codec is still used.

flouwers, 2018-02-27:
Can you please tell me how do I get the bug fix ? I ran a self-update, followed by a configure, but omxmjpegdec is still used.

fnoop, 2018-02-27:
Sorry, doing a self-update will only update the stable branch which is pretty frozen now as we're in heavy development for the next major 2.0 update.  You can change to the master branch (http://goodrobots.github.io/maverick/#/?id=maverick-code-branch), but it's not well tested at the moment as there is so much development going on.

You're better off just hacking the visiond file directly - /srv/maverick/software/maverick/manifests/maverick-modules/maverick_vision/files/maverick-visiond
Look for the line:
`if Gst.ElementFactory.find("omxmjpegdec"):`
and change it to
`if Gst.ElementFactory.find("omxmjpegdecDISABLED"):`
Then restart visiond
`maverick restart visiond`

flouwers, 2018-02-27:
Perfect! 👍  That works fine now with automatic pipeline!

closed 2018-02-27 by flouwers

fnoop, 2018-02-27:
Great :)  Thanks for the testing and feedback :)

meppe, 2019-03-15:
Hi,
with Maverick 1.1.6 the changes suggested by @fnoop are already there. However, I still don't get a send-receive setup with various configurations. 

@flouwers, could you please also tell which pipeline you use for receiving the video? Since I cannot receive anything with VLC on my Windows PC, I tried videosink on the same RP0 on which I am sending the video, but this does not work either.

For example, I followed your suggestion above and used the following pipeline for sending:

`gst-launch-1.0 -v v4l2src device=/dev/video0 ! 'video/x-raw, width=640, height=480, framerate=60/1' ! queue ! autovideoconvert ! omxh264enc ! h264parse ! rtph264pay ! udpsink host=127.0.0.1 port=5600`

To receive the video I use the following pipeline: (and have tried countless others...)

`gst-launch-1.0 -v -e udpsrc port=5600 ! rtph264depay ! omxh264dec ! avimux ! filesink location=video/test.avi`

However, this results in an avi file which is on 1kb and not viewable.

Given the above pipeline for sending, and entering the IP of my Windows PC as udpsink host, for watching the file via VLC does not help either. What do I have to enter exacly when opening the network stream in VLC? 

Of course, not to mention that the usual maverick start visiond script does not work at all. The camera does not even get activated (red LED not on). This is the log when I start visiond via maverick using the automatic pipeline:

2019-03-15 19:49:44,486 - INFO - Starting maverick-visiond
2019-03-15 19:49:44,518 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2019-03-15 19:49:44,547 - DEBUG - V4l2 device input: Camera 0:2
2019-03-15 19:49:44,549 - INFO - v4l2 device /dev/video0 is a camera, autoselecting
2019-03-15 19:49:44,561 - DEBUG - V4l2 device input: Camera 0:2
2019-03-15 19:49:44,563 - DEBUG - driver: bm2835 mmal
2019-03-15 19:49:44,565 - DEBUG - card: mmal service 16.1
2019-03-15 19:49:44,570 - DEBUG - Camera control: Brightness
2019-03-15 19:49:44,576 - DEBUG - Camera control: Contrast
2019-03-15 19:49:44,578 - DEBUG - Camera control: Saturation
2019-03-15 19:49:44,586 - DEBUG - Camera control: Red Balance
2019-03-15 19:49:44,588 - DEBUG - Camera control: Blue Balance
2019-03-15 19:49:44,597 - DEBUG - Camera control: Horizontal Flip
2019-03-15 19:49:44,599 - DEBUG - Camera control: Vertical Flip
2019-03-15 19:49:44,607 - DEBUG - Camera control: Power Line Frequency
2019-03-15 19:49:44,614 - DEBUG - Camera control: Sharpness
2019-03-15 19:49:44,616 - DEBUG - Camera control: Color Effects
2019-03-15 19:49:44,618 - DEBUG - Camera control: Rotate
2019-03-15 19:49:44,629 - DEBUG - Camera format: Planar YUV 4:2:0
2019-03-15 19:49:44,631 - DEBUG - Camera format: YUYV 4:2:2
2019-03-15 19:49:44,633 - DEBUG - Camera format: 24-bit RGB 8-8-8
2019-03-15 19:49:44,644 - DEBUG - Camera format: JFIF JPEG
2019-03-15 19:49:44,645 - DEBUG - Camera format: H.264
2019-03-15 19:49:44,647 - DEBUG - Camera format: Motion-JPEG
2019-03-15 19:49:44,648 - DEBUG - Camera format: YVYU 4:2:2
2019-03-15 19:49:44,661 - DEBUG - Camera format: VYUY 4:2:2
2019-03-15 19:49:44,663 - DEBUG - Camera format: UYVY 4:2:2
2019-03-15 19:49:44,664 - DEBUG - Camera format: Y/CbCr 4:2:0
2019-03-15 19:49:44,674 - DEBUG - Camera format: 24-bit BGR 8-8-8
2019-03-15 19:49:44,676 - DEBUG - Camera format: Planar YVU 4:2:0
2019-03-15 19:49:44,682 - DEBUG - Camera format: Y/CrCb 4:2:0
2019-03-15 19:49:44,684 - DEBUG - Camera format: 32-bit BGRA/X 8-8-8-8
2019-03-15 19:49:44,707 - INFO - Camera YUV stream available, using yuv stream
2019-03-15 19:49:44,712 - INFO - Raspberry Pi Camera detected, setting pixel format to I420
2019-03-15 19:49:44,713 - INFO - Creating stream object - camera:/dev/video0, stream:yuv, pixelformat:I420, encoder:h264, size:(1280 x 720 / 30), output:rtsp, brightness:0
2019-03-15 19:49:44,715 - INFO - Attaching input 'v4l2': /dev/video0
2019-03-15 19:49:44,886 - INFO - Attaching stream 'yuv'
2019-03-15 19:49:44,978 - INFO - Attaching encoding 'h264'
2019-03-15 19:49:44,991 - INFO - Raspberry hardware encoder detected, using omxh264enc as h264 encoder
2019-03-15 19:49:46,945 - INFO - Attaching payload 'h264'
2019-03-15 19:49:46,987 - DEBUG - h264parse element created
2019-03-15 19:49:47,049 - DEBUG - Attaching h264pay to h264parse
2019-03-15 19:49:47,071 - INFO - Attaching output 'rtsp'
2019-03-15 19:49:47,131 - INFO - RTSP stream running at rtsp://0.0.0.0:8554/video
2019-03-15 19:49:47,159 - INFO - Pipeline: "v4l2-source /dev/video0 ! capsfilter 'video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, framerate=(fraction)30/1' ! queue ! autovideoconvert ! raspberry-h264-encode ! x264capsfilter ! h264parse ! pay0"
2019-03-15 19:49:47,172 - INFO - Starting camera stream

I am fighting this for a week now... any help appreciated!

fnoop, 2019-03-15:
Hi @meppe sorry you've been banging your head against the wall!  gstreamer can be super frustrating sometimes.  Would suggest you head over to our gitter room (https://gitter.im/goodrobots/maverick) for some interactive help.

Couple of things to note:
 - Try and capture h264 stream to start with, rather than raw stream and then pass through the omx encoder (if specifying a manual pipeline)
 - This issue was about USB cameras which are very different from the raspberry camera
 - The automatic pipeline should work with raspberry, it's been fairly well tested.  So with the automatic pipeline and the log above, you don't get a red light on the camera?
 - The default pipeline above is RTSP - are you using VLC rtsp URL to connect? https://goodrobots.github.io/maverick/current/#/modules/vision?id=vlc-rtsp