goodrobots/maverick #737: Add SITL support for ArduPlane

SamuelDudley, 2018-03-06:
Add JSBSim to maverick and add option to launch SITL interface as a plane

fnoop, 2018-03-06:
http://goodrobots.github.io/maverick/#/modules/dev?id=ardupilot-firmware

fnoop, 2018-03-06:
TL;DR: Set `"maverick_dev::ardupilot::ardupilot_vehicle": "plane"` in localconf

SamuelDudley, 2018-03-06:
I'm happy to help out with supporting this one. Might be a good first effort into adding features into maverick via puppet. 

https://github.com/tridge/jsbsim

fnoop, 2018-03-06:
Sorry I mean it's already implemented :)  All you have to do is set the above localconf param and run `maverick configure`.  It should compile, install and configure jsbsim and plane firmware for you

SamuelDudley, 2018-03-06:
Is it using the JSBSim FDM backend as linked above? Or the python Sim?

fnoop, 2018-03-06:
https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/maverick_dev/manifests/sitl.pp#L3
It's been a while since I've tried it, we should try setting it up and see how it goes.

SamuelDudley, 2018-03-06:
Ah, sorry on my phone at the moment! Looks like you have it covered :) I pretty much exclusively use JSBSim for plane SITL so I'll see how it goes and report back what I find.

SamuelDudley, 2018-03-07:
Closing for now. If something is broken I will re open.

SamuelDudley, 2018-03-07:
Just tried setting up plane on the dev VM
```
Maverick Environment:    dev 
Proceeding to update system configuration - please be patient, this can take a while..

Warning: Scope(Maverick_dev::Fwbuildwaf[sitl_plane]): Ardupilot Firmware: plane will be compiled and can take a while, please be patient
Notice: Scope(Oncevcsrepo[git-jsbsim]): oncevcsrepo: /srv/maverick/var/build/jsbsim git repo doesn't exist locally, cloning may take a while..
Notice: Compiled catalog for goodrobots-maverick in environment dev in 5.95 seconds
Notice: /Stage[main]/Maverick_dev::Sitl/File[/srv/maverick/config/dev/sitl-vehicle.conf]/content: content changed '{md5}ea5ca6929dbf0afaf69ec458d18dd626' to '{md5}28e9a1e1293a6d315d3a074dc3880177'
Notice: /Stage[main]/Maverick_dev::Ardupilot/Maverick_dev::Fwbuildwaf[sitl_plane]/Exec[ardupilotfw_sitl_plane]/returns: executed successfully
Notice: /Stage[main]/Maverick_dev::Sitl/Oncevcsrepo[git-jsbsim]/File[/srv/maverick/var/build/jsbsim]/ensure: created
Error: Could not find command '/srv/maverick/var/build/jsbsim/autogen.sh'
Error: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-autogen]/returns: change from 'notrun' to ['0'] failed: Could not find command '/srv/maverick/var/build/jsbsim/autogen.sh'
Notice: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-make]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-make]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-makeinstall]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-makeinstall]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/File[/srv/maverick/software/jsbsim/bin]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/File[/srv/maverick/software/jsbsim/bin]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-cpbin]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-cpbin]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/File[/srv/maverick/var/build/.install_flag_jsbsim]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/File[/srv/maverick/var/build/.install_flag_jsbsim]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/Oncevcsrepo[git-jsbsim]/Vcsrepo[/srv/maverick/var/build/jsbsim]/ensure: created
Notice: /Stage[main]/Maverick_dev::Sitl/Service_wrapper[maverick-sitl]/Service[maverick-sitl]: Triggered 'refresh' from 1 event
Notice: Applied catalog in 120.05 seconds

Maverick finished, happy flying :)
```

fnoop, 2018-03-07:
Ok this was just an ordering issue with the manifest - forcing order in a declarative/non-deterministic is tricky.  If you get errors when running maverick configure, try running it again and often it will go through.
For this specific issue, if you look carefully at the output you'll see the git repo is only pulled at the end, after it tries to build.  The build fails because the repo hasn't been pulled yet:
```
Notice: /Stage[main]/Maverick_dev::Sitl/Oncevcsrepo[git-jsbsim]/Vcsrepo[/srv/maverick/var/build/jsbsim]/ensure: created
```
^^ This comes after:
```
Error: Could not find command '/srv/maverick/var/build/jsbsim/autogen.sh'
Error: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-autogen]/returns: change from 'notrun' to ['0'] failed: Could not find command '/srv/maverick/var/build/jsbsim/autogen.sh'
Notice: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-make]: Dependency Exec[jsbsim-autogen] has failures: true
Warning: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-make]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Sitl/Exec[jsbsim-makeinstall]: Dependency Exec[jsbsim-autogen] has failures: true
```

commit 1213f48e (2018-03-07): Fix ardupilot sitl/jsbsim ordering issue, #fixes #737
closed 2018-03-07 by fnoop