goodrobots/maverick #738: Have Maverick send MAVlink messages from SITL out, via telemetry radios

Rusty105, 2018-03-21:
Ok, some background first:

I have installed Maverick on a Raspberry Pi 3 and have SITL running a instance of 'copter'. I can connect to SITL via TCP in Mission Planner and have it run virtual missions. As far as I can tell Mission Planner thinks it is a real copter. All good so far. 

I am also in the process of building an antenna tracker, using Ardutracker and a Pixhawk, on a 3D printed pan/tilt platform I am designing. What I would like to have happen is that MAVLink messages would be sent to the Pixhawk on the tracker from the Raspberry Pi3 via telemetry radios. This would allow me to test the mechanical design of the tracker.

As discussed in the Ardupilot forum,  https://discuss.ardupilot.org/t/using-a-raspi3-to-run-sitl-can-it-be-done/24099  post 18, it appears that Maverick only receives MAVLink messages via telemetry radios.

SO, what I am asking as a feature is the ability for Maverick to be able to SEND MAVLink messages over a connected telemetry radio. I think having this feature would be a good addition to Maverick, it would allow users to use Maverick to more fully imitate a UAV and allow it to be used to test other components in the system via telemetry radios.

If I somehow missed where this can be accomplished with the existing feature set, please let me know.

Rusty

fnoop, 2018-03-25:
Interesting question.  Theoretically this can possibly be done at the moment thanks to mavlink routing by setting the usb as the IN device.  I think depending on the router you should still be able to connect to it and it will act as a 'sender'.  I'll test with a set of radios.
Alternatively, a separate serial endpoint option can be added.

fnoop, 2018-03-25:
BTW, if you had any interest we would love to have a tracker recipe in goodrobots - an easy to follow procedure to setup the hardware and software.  It's a model I see being a very popular deployment model for the upcoming maverick 2.0 release.

Rusty105, 2018-03-27:
Following up. 
I edited the localconfig.json (sp?) file and un commented the line the specified MAVPROXY as the proxy to use, did a "maverick configure" and a reboot and no change,, all the while the tracker was listening for a UAV on the telemetry radio. I then entered "mavproxy.py --out dev/tty/USB0"  and the tracker sprang to life, however I do not think it was receiving correct information, it was randomly pointing in many different directions, and I was seeing random characters in the terminal window from where I had issued the mavproxy.py command. I am going to try and debug with an actual UAV and see if the tracker is working correctly before I go further.

Rusty105, 2018-03-27:
RE:  "Tracker Recipe"  Interested, but need more info...   What are you looking for ? I am currently designing/3D printing a tracker, using Ardutracker as the brains, I plan on documenting that build for my website/blog, and could share that??

fnoop, 2018-03-27:
Central logic support for outbound serial device added, and mavlink-router config template updated and tested.
Still have to add support for cmavnode and mavproxy.

fnoop, 2018-03-28:
re: tracker if you're interested come and have a chat in gitter: https://gitter.im/goodrobots/Lobby

fnoop, 2018-03-29:
Confirmed working with two raspberries and set of 3drv2 radios: on a raspberry pi 3, set sitl to forward mavlink out of usb/radio:
```
	"maverick_dev::sitl::mavlink_serialout": "/dev/ttyUSB0",
	"maverick_dev::sitl::mavlink_outbaud": 115200,
```
On raspberry pi zero with other radio, set to receive from usb:
```
	"maverick_fc::mavlink_input": "/dev/ttyUSB0", // Input source for mavlink proxy, how the Companion Computer connects to the Flight Controller
	"maverick_fc::mavlink_baud": "115200", // Input baud rate for UART connections
```

closed 2018-03-29 by fnoop