goodrobots/maverick #740: SITL arduplane doesn’t work

fnoop, 2018-03-23:
Setting ArduPlane in ~/config/dev/sitl-vehicle.conf and FRAME=quadplane in ~/configdev/sitl.conf compiles and runs arduplane, but no radio/sensors are configured or calibrated so can't fly.

Rusty105, 2018-03-27:
Was going to mention this, but thought it was just me. I was able to switch the SITL model to "plane" but it required 3D accel calibration.

fnoop, 2018-03-29:
OK so this has to do with plane time.  If you switch to ArduPlane in sitl-vehicle.conf and FRAME=quadplane in ~/config/dev/sitl.conf and set WIPE_EEPROM=true, then restart sitl, then set WIPE_EEPROM=false and restart sitl, then you get calibrated sensors and can take off and fly.
If you change FRAME=plane and wipe the eeprom, you don't get calibrated sensors, and jsbsim doesn't run.  Need to fix that.

fnoop, 2018-03-29:
Ah so the plane model has been changed to jsbsim, so set:
FRAME=jsbsim
After wiping eeprom, this fires up the sim_vehicle and the JSBSim, and reports to connect the two together but then a lot of these:
`5: GEARerror recv on control socket: Bad file descriptor_CONTACT: TAIL_LG 0`
And still no calibrated sensors.

fnoop, 2018-03-29:
after restarting sitl services and wiping eeprom, this works now.  I don't think it's going to be possible to automate this, so this will be 'fixed' by better documentation.

fnoop, 2018-03-29:
Note: bad ahrs/gps messages when trying to arm are just the EKF settling.  Wait a minute or two and try again.

closed 2019-03-27 by fnoop