goodrobots/maverick #757: Create Minimal Config For Digital FPV

cglusky, 2018-04-21:
//---Web Services---

//Cloud 9
  "maverick_cloud9::cloud9_active": false,

//---Linux Desktop---

//Desktop
  "maverick_desktop::enable": false,

//---Dev---

//ArduPilot SITL
  "maverick_dev::sitl:sitl_active": false,

//Mavlink SITL Proxy
  "maverick_dev::sitl::mavlink_active": false,

//ROS
  "maverick_dev::sitl::ros_instance": false,

//ROS Master
  "maverick_dev::sitl::rosmaster_active": false,

//MAVROS
  "maverick_dev::sitl::mavros_active": false,

//---Flight Controller---

//Mavlink Proxy
  "maverick_fc::mavlink_active": false,
  "maverick_fc::mavlink_input": "/dev/ttyAMA0",
  "maverick_fc::mavlink_baud": 115200,

//ROS
  "maverick_fc::ros_instance": false,
  "maverick_fc::rosmaster_active": false,

//MAVROS
  "maverick_fc::mavros_active": false,

//---Mavlink---

//MAVCesium
  "maverick_mavlink::mavcesium_active": false,

//---Network---

//ZeroTier
  "maverick_network::zerotier::active": false,

//---Vision---

//Visiond
  "maverick_vision::visiond::active": true,

//WebVision
  "maverick_vision::webvision_active": false,

//Seek
  "maverick_vision::vision_seek::active": false,

//Vision Landing
  "maverick_vision::vision_landing::active": false,

//Camera Streaming Daemon
  "maverick_vision::camera_streaming_daemon::active": false,

//---Hardware---

//FLIROne
  "maverick_hardware::flirone::active": false,

cglusky, 2018-04-21:
waiting for it to finish initial config run on raspi0.  will also need to add config for wifibroadcast.

and doc visiond setup.

cglusky, 2018-04-21:
also depends on #756 for some web services control.  meantime can just comment those out and stop manually if an issue.

fnoop, 2018-04-22:
So this really depends on what you're trying to accomplish.  The raspberrylite profile is intended to run on the pi zero as it's target platform, with enough gpu memory allocated for video.  Running digital fpv on the raspberry barely takes any cpu power or memory however because it's all done within the gpu/vpu, so there's no need to shut everything else off.
In addition on a very resource constrained platform it's nice to be able to see what is happening to resources and where your bottlenecks are, which is why I left the analysis stuff running.
When maverick-web is up and running (shortly) then this will be a fundamental service so then ros/mavros, api and nginx service needs to be running.
We could create a raspberrylite-minimal profile for those specifically want nothing else to be running, but I'm not sure I see the point.  What's the problem with the current profile?  Are you trying to solve a problem that really exists?

fnoop, 2018-04-22:
Raised #758 to add control of nginx

cglusky, 2018-04-22:
I am not sure it should be supported.  But if we are going to doc a digital FPV build then I would like it to work well.  Fair or not, the first question will be "Why not just use ezwifibroadcast?"  If Maverick does not provide real value in this use case we should not force it.  Thought I could check it out and see so we know; currently this is what I have:

- 8.63 hrs to run maverick config the first time.  With lots of errors. 
- Config settings are ignored
- Taking minutes to boot

Right now my lack of knowledge is producing a wood stove in a microwave world.

[maverick.flight.2018-04-21-23-51.log](https://github.com/goodrobots/maverick/files/1935582/maverick.flight.2018-04-21-23-51.log)

cglusky, 2018-04-22:
easier just to use command line and disable whatever services.  will document once i get time to test.

fnoop, 2018-04-22:
```
sudo hostnamectl set-hostname maverick-raspberrylite-minimal
maverick configure
```

fnoop, 2018-04-22:
Even though rosmaster@fc disabled, still trying to start:
```
Apr 22 23:31:44 maverick-raspberrylite-minimal systemd[1]: maverick-rosmaster@Fc.service: Service hold-off time over, scheduling restart.
Apr 22 23:31:44 maverick-raspberrylite-minimal systemd[1]: Stopped Maverick ROS Master - Fc.
Apr 22 23:31:44 maverick-raspberrylite-minimal systemd[1]: Started Maverick ROS Master - Fc.
```

fnoop, 2018-04-23:
ros still trying to start due to rogue localconf.json entries, with default localconf.json, everything stopped with minimal config.

fnoop, 2018-04-23:
Turning off the maverick-uploader service really reduces load and memory usage on pi zero to almost zero

fnoop, 2018-04-24:
Turn analysis services off by default in raspberrylite profile.

fnoop, 2018-04-25:
Updated raspberrylite profile with analysis turned off is a very good compromise between the two:
```
Maverick Environment:    flight

Service                       Description               Status | At Boot
------------                  ------------             ------------------
Flight Controller Services
mavlink-router@fc             Mavlink (FC)             Running | Enabled
rosmaster@fc                  ROS (FC)                 Running | Enabled
mavros@fc                     Mavros (FC)              Running | Enabled
api@fc                        API (FC)                 Stopped | Disabled
------------                  ------------             ------------------
Web Services
webdev                        Maverick Web Devserver   Stopped | Disabled
nginx                         Webserver                Running | Enabled
cloud9                        Cloud9 IDE               Running | Enabled
mavcesium                     MavCesium                Stopped | Disabled
webvision                     Web Vision               Stopped | Disabled
------------                  ------------             ------------------
Analysis Services
grafana                       Analysis Dashboard       Stopped | Disabled
mavlogd                       Log Import Daemon        Stopped | Disabled
influxd                       Time Data DB             Stopped | Disabled
collectd                      System Metrics           Stopped | Disabled
uploader                      Log Uploader Daemon      Stopped | Disabled
------------                  ------------             ------------------
Vision Services
visiond                       Vision Daemon            Stopped | Disabled
csd                           Camera Streaming Daemon  Stopped | Disabled
vision_landing                Vision Landing           Stopped | Disabled
vision_seek                   Seek Thermal Vision      Stopped | Disabled
flirone                       Flir One Vision          Stopped | Disabled
coav                          Collision Avoidance      Stopped | Disabled
```

After boot on a pi zero, load is fairly flat:
```
[flight] [mav@maverick-raspberrylite ~/software/maverick]$ top
top - 07:31:19 up 15 min,  2 users,  load average: 0.11, 0.23, 0.27
Tasks:  82 total,   1 running,  55 sleeping,   0 stopped,   0 zombie
%Cpu(s): 12.9 us,  5.8 sy,  0.0 ni, 80.3 id,  0.7 wa,  0.0 hi,  0.2 si,  0.0 st
KiB Mem :   378940 total,    68084 free,   148884 used,   161972 buff/cache
KiB Swap:   102396 total,   100860 free,     1536 used.   174968 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
 1890 mav       20   0    8064   3268   2760 R 15.8  0.9   0:00.11 top
  730 mav       20   0   58760  29004   6632 S  5.3  7.7   0:13.50 roscore
  817 mav       20   0   58816  29304   6684 S  5.3  7.7   0:14.71 roslaunch
 1032 mav       20   0  166192  37336  16792 S  5.3  9.9   0:18.06 mavros_node
    1 root      20   0    9600   5948   4840 S  0.0  1.6   0:05.10 systemd
    2 root      20   0       0      0      0 S  0.0  0.0   0:00.00 kthreadd
    4 root       0 -20       0      0      0 I  0.0  0.0   0:00.00 kworker/0:0H
    6 root       0 -20       0      0      0 I  0.0  0.0   0:00.00 mm_percpu_wq
    7 root      20   0       0      0      0 S  0.0  0.0   0:00.85 ksoftirqd/0
```
    
This leaves cloud9 and ROS/MAVROS running as well as nginx webserver (laying groundwork for future maverick-web).  I don't see any need for raspberrylite-minimal but will leave in place.

closed 2018-04-25 by fnoop