goodrobots/maverick #759: Add built maverick-web instance and nginx locations

fnoop, 2018-04-22:
Maverick web is currently available for devserver in /dev/maverick and if it exists in dev repo, /web/maverick.
Instead add a 'dist' repo for maverick-web and repoint /web/maverick at that.

fnoop, 2020-02-27:
Update a dist/ directory in maverick-web, or create a maverick-web-dist repo?

SamuelDudley, 2020-03-01:
Regards of if you want the dist in maverick-web or in another repo we can automate the build in a GitHub pipeline. E.g. push maverick-web triggers a dist build that either is available as a release download (that we can pull down into maverick) or pushes the result of the dist build to maverick-web-dist.

fnoop, 2020-03-01:
I think at least to start with we would want this to be a manual process.  The dist should be a well tested, stable codebase, probably tagged in github, that shouldn't be updated that often.

We could use the pipeline by setting releases eventually, once we're a bit further down the line.

fnoop, 2020-03-24:
This is working really well now.  Dev is done in ~/code/maverick-web and any pushes are automatically built and pushed to -web-dist repo which lives in ~/software/maverick-web. `maverick self-update` updates both repos.
What's left to do?
- Add a stable branch to both and make that the default for -web-dist, so we don't break users interface while we develop.
 - Make -web-dist the default web interface for Maverick, and retire -web-legacy.

fnoop, 2020-03-29:
-web-dist is now the default website for maverick and the autopush from -web is working perfectly, great solution.
Need to now create -stable branches and point maverick webroot at -stable branch.

fnoop, 2020-04-26:
- Document
- Retire web-legacy

fnoop, 2020-04-26:
-web-legacy addressed in #955