goodrobots/maverick #760: Boot time slow on raspberry

fnoop, 2018-04-22:
Profile and improve boot time where possible

fnoop, 2018-04-22:
@cglusky Boot time is a complex issue.  Modern unix systems that use systemd parallelize system service startup by compiling a dependency graph, and there are certain milestones that are used to order groups of services.
It may take a while to fully boot but you can try and order the user services as much up front that will be most useful -eg. video, mavlink-router.

There's a very useful tool: systemd-analyze.  Running this without arugments on a freshly installed maverick-raspberrylite on a pi zero gives:
```
[flight] [mav@maverick-raspberrylite ~]$ systemd-analyze
Startup finished in 2.325s (kernel) + 38.062s (userspace) = 40.388s
```
That's 40 seconds in total for all services, which isn't too bad.  You can 'blame' to find the biggest hitters:
```
[flight] [mav@maverick-raspberrylite ~]$ systemd-analyze blame
         16.468s maverick-motd.service
          9.426s networking.service
          7.460s dev-mmcblk0p2.device
          3.880s keyboard-setup.service
          2.107s dphys-swapfile.service
          2.018s systemd-logind.service
          1.578s rfkill-unblock.service
          1.519s systemd-udev-trigger.service
          1.463s wifi-country.service
          1.419s ssh.service
          1.418s systemd-fsck-root.service
          1.349s netfilter-persistent.service
          1.318s systemd-journald.service
          1.261s raspi-config.service
          1.181s avahi-daemon.service
          1.139s binfmt-support.service
          1.082s resolvconf.service
          1.054s systemd-udevd.service
          1.021s dev-mqueue.mount
```

So that's interesting, a lot of that can be improved, in particular maverick-motd.

`systemd-analyze plot` is also very useful, it gives a visual graph that can be used to easily see where the slowest services are:

[systemd-analyze.plot.svg](https://shipusercontent.com/29b9734f5a36be8fb573faafe9fce9e3/systemd-analyze.plot.svg)

fnoop, 2018-04-23:
[flight] [mav@maverick-raspberrylite-minimal ~]$ systemd-analyze
Startup finished in 1.831s (kernel) + 30.365s (userspace) = 32.196s

[boot.svg](https://shipusercontent.com/eac1a1b8693e5c7fc4000c7b5c3909c8/boot.svg)

```
          9.909s networking.service
          8.582s maverick-motd.service
          7.590s dev-mmcblk0p2.device
          4.109s keyboard-setup.service
          2.307s dphys-swapfile.service
          2.217s raspi-config.service
          1.716s systemd-udev-trigger.service
          1.692s binfmt-support.service
          1.554s wifi-country.service
          1.449s triggerhappy.service
          1.436s resolvconf.service
          1.404s netfilter-persistent.service
          1.338s rfkill-unblock.service
          1.300s systemd-fsck-root.service
          1.289s kmod-static-nodes.service
          1.282s systemd-journald.service
          1.279s rsyslog.service
          1.247s fake-hwclock.service
          1.152s systemd-udevd.service
          1.069s alsa-restore.service
```

maverick-motd needs to be improved if possible.  This is a dependency of multi-user target milestone, so affects boot time to ssh login significantly.

fnoop, 2018-04-24:
maverick-motd is slow because it is calling maverick-info.  maverick-info is slow because it is calling a complete facter run and then parsing the facts and storing as more facts, as part of the Baseos class.  The only thing that uses Baseos is maverick-info, so this should be rewritten to not call facter but instead get the results from a python module or direct from the OS.  This will save 8secs or a full quarter of the boot time.

fnoop, 2018-04-24:
Difficult to replace facter, will take a lot of effort with python alone.  Instead, don't make multi-user or anything else dependent on maverick-motd.service at boot.  This will mean someone logging in before motd has run fully will get slightly out of date banner, but otherwise will cut off 25% off the boottime on raspberries.

cglusky, 2018-04-26:
Appreciate the work on this.

fnoop, 2018-04-27:
Moved rosmaster service to start after network-online, not maverick-motd which takes ages to start.  

@cglusky Startup plot for an up core attached.  Note that the system is essentially booted after 4.5seconds.  It then proceeds with the network system startup and takes forever (a further 8 seconds) to associate with the wireless router, and only after that it can start network services that depend on the network (wireless) interface, which take a further 4 seconds to initiate, for a total 16 seconds boot time.
Compare that with a stripped out dedicated OS for wifibroadcast, eg. ezwifibroadcast - this only has to boot the core system (4.5 seconds in the above example), and then there is no time necessary to connect/associate with a router as the wifibroadcast is connectionless.  It can immediately start firing out network packets.  This is why you perceive it as a much faster startup time, and should be similar in Maverick when configured for wifibroadcast.

cglusky, 2018-04-27:
Very nice.  And that makes a bunch of sense.  I'll do another self-update to make sure I am running latest on the D300.

Concur on the minimal but not might hurt to keep it around.  You may find that it falls into the nylon/aluminium fastener category...Makes people feel faster...Like wearing UnderArmour.

closed 2018-05-28 by fnoop