goodrobots/maverick #762: Flashing Error for TX2

smafaith, 2018-04-24:
After getting tx2 to recovery mode, I run "sudo ./flash.sh -r jetson-tx2 mmcblk0p1" and it goes like this in terminal:
"Error: Return value 8
Command tegrarcm_v2 --chip 0x18 --rcm rcm_list_signed.xml --skipuid
Reading board information failed." 
Failing from the beginning. Help!

smafaith, 2018-04-24:
For the first running, the error information is: "UnicodeDecodeError: 'ascii' codec can't decode byte 0xe4 in position 0: ordinal not in range(128) Reading board information failed." 
And for retrying, the error goes like that above.

fnoop, 2018-04-24:
Hey, thanks for the report.  Unfortunately I don't have a TX2 so put this together somewhat blind, but I have had a report of at least one person getting it to work.
It's on my todo list to update the images to be based around jetpack 3.2, I'll try and do that tonight and get new images uploaded.

smafaith, 2018-04-24:
It's quite interesting because I can get different error messages for different trying. Latest trying goes further with the error of "Reusing existing system.img... file does not exist." 
Thank you fnoop.

fnoop, 2018-04-24:
That is weird!  Are you using the dev board or another carrier?

smafaith, 2018-04-24:
I'm using the dev board. And I've installed Jetpack 3.0.

smafaith, 2018-04-24:
Okay, I'll have a try. Thank you fnoop.

smafaith, 2018-04-25:
Dear fnoop, I failed in updating to jetpack 3.1 and 3.2 for bad network connection. 
I have two questions: 
1. Will network connection affect maverick installation? 
2. I unzip the tgz on my Ubuntu 16.04 laptop and then try to flash my tx2-dev with "sudo ./flash.sh -r jetson-tx2 mmcblk0p1", are the installation steps right?

smafaith, 2018-04-26:
Yes, I followed the instructions, and I think the download of tgz is good. My Ubuntu system is independent, not running in a VM. Maybe I should try on a new computer.

smafaith, 2018-04-26:
I followed the guide to try bootstrap on tx2 and it went like this: 
"Maverick - UAV Companion Computer System - Version 1.1.6

Environment marker set but is being ignored as --env is set
Maverick Environment:    dev 
Proceeding to update system configuration - please be patient, this can take a while..

Warning: Facter: Could not retrieve fact='pip_version', resolution='<anonymous>': undefined method `[]' for nil:NilClass
Warning: Scope(Class[Maverick_web::Cloud9]): Cloud9 will be compiled and can take a long time, please be patient..
Error: Evaluation Error: Error while evaluating a Resource Statement, Duplicate declaration: Oncevcsrepo[git-aruco] is already declared at (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_hardware/manifests/tegra.pp, line: 12); cannot redeclare (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_vision/manifests/aruco.pp, line: 9) (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_vision/manifests/aruco.pp, line: 9, column: 9) on node tegra-ubuntu

Maverick finished, happy flying :)"

fnoop, 2018-04-26:
The TX1 image has a pre-squashed boot/system.img.  The TX2 doesn't but it should generate one, although it needs a lot of space to do so.  I'll look and see if I can make a squashed image for TX2 as well.

fnoop, 2018-04-26:
OK, could you please try this command?
```sudo ./flash.sh jetson-tx2 mmcblk0p1```

If that doesn't work, try:
```sudo ./flash.sh -R rootfs jetson-tx2 mmcblk0p1```

You'll need quite a bit of disk space as it needs to build system.img.

fnoop, 2018-04-26:
Need to document tegra image creation process:
https://github.com/goodrobots/maverick/issues/706

smafaith, 2018-04-30:
Good job fnoop, it works. Thank you so much!

fnoop, 2018-04-30:
Fab!  Did it take a lot of space creating the system.img before flashing?

smafaith, 2018-04-30:
Yes, it did take a lot of space and may lead to a problem of "The system is running in low-graphics mode". So I have to flash it back to Jetpack 3.1...

fnoop, 2018-04-30:
Which command worked in the end?
Would you be interested in the uploading the system.img, so I can create a squashed download image for the tx2?  If the squashed system.img is provided in the flash image then it doesn't need all that space to recreate it, and it's a lot faster.

smafaith, 2018-04-30:
" sudo ./flash.sh -R rootfs jetson-tx2 mmcblk0p1" This worked at last. I've kept the " system.img" file and how can I upload that? And it turned out to be failed at last for the "low-graphics mode" issue.

smafaith, 2018-05-02:
Hello fnoop, shall I use ssh or Web to run "maverick" codes on tx2 after flashing or connect the keyboard, mouse, and screen to run the codes in a terminal? If I should use ssh, then what is the local host I need to visit? If I should use the terminal on tx2, how to solve the problem of "The system is running in low-graphics mode"?

fnoop, 2018-05-02:
Hey @smafaith did you unpack the maverick image as root, or through sudo?  Or did you run out of disk space while flashing (while it was creating the system.img)?  Have a read here:
https://devtalk.nvidia.com/default/topic/1027037/jetson-tx2/unable-to-flash-tx2-with-jetpack-3-1-gt-quot-the-system-is-running-in-low-graphics-mode-quot-/

fnoop, 2018-05-02:
The easiest way to get going is to plug an ethernet cable into the dev board, and then ssh to 'maverick-tegra.local' (if you're using a mac or linux computer, otherwise you'll have to find the IP from your router or otherways).  You can also go to maverick-tegra.local in a browser and get some temporary web links - the real web interface is not until the upcoming 2.0 version.  But the Cloud9 interface is very useful: http://maverick-tegra.local:6789

smafaith, 2018-05-03:
Good news dear fnoop, I've visited and self-updated maverick on Cloud9, and after self-update, the system runs well. So I think the problem of "The system is running in low-graphics mode" is probably from an unsupported HDMI cable. Besides, I run `sudo ./flash.sh jetson-tx2 mmcblk0p1` this time and it is able to generate "system.img" and flash it on TX2, and this code takes a shorter time than `sudo ./flash.sh -R rootfs jetson-tx2 mmcblk0p1`. 
What's more, I can't find the connection introductions for Maverick-TX2 and Pixhawk on either maverick pages or Ardupilot pages, so I wonder for a tx2 dev board, do I have to connect a J120 carrier board as middleware or connector? Thank you.

fnoop, 2018-05-03:
@smafaith Great!  I'll update the docs.
No idea about connecting pixhawk to the dev board, I'll have a look at update the docs.
It would be a good idea for the quick start for each hardware type, to then follow on to a 'what's next' section of the docs.  I'll label this issue for documentation.
Thanks for your efforts and feedback!

fnoop, 2018-05-07:
@smafaith I've never used vision_landing on the tegra platform - I don't think CUDA actually provides accelerated functions for most of the basic hard work that is needed in vision_landing, and it's all done on the CPU which isn't the strong point of the tegra (it's essentially a midrange graphics card).  Also you should read the warning on the main github page: https://github.com/goodrobots/vision_landing

Neither ardupilot or px4 currently support vision based precision landing yet, at least not without some indepth knowledge and a bit of extra work.  You can try it but depending on the performance you get, it may not be reliable/safe.

Anyway to answer your questions:
 1. I don't know about the dev board, it's not very useful for actually flying unless you have a gigantic UAV so never tried it.  But yes you should be able to, and it would be very useful to document somewhere.
2. vision_landing doesn't use MAVROS or ROS so you can disable those.