goodrobots/maverick #770: Network: networking service errors

fnoop, 2018-05-01:
```
May 01 12:04:43 maverick-up systemd[1]: Starting Raise network interfaces...
May 01 12:04:44 maverick-up dhclient[598]: Internet Systems Consortium DHCP Client 4.3.5
May 01 12:04:44 maverick-up ifup[448]: Internet Systems Consortium DHCP Client 4.3.5
May 01 12:04:44 maverick-up ifup[448]: Copyright 2004-2016 Internet Systems Consortium.
May 01 12:04:44 maverick-up ifup[448]: All rights reserved.
May 01 12:04:44 maverick-up ifup[448]: For info, please visit https://www.isc.org/software/dhcp/
May 01 12:04:44 maverick-up dhclient[598]: Copyright 2004-2016 Internet Systems Consortium.
May 01 12:04:44 maverick-up dhclient[598]: All rights reserved.
May 01 12:04:44 maverick-up dhclient[598]: For info, please visit https://www.isc.org/software/dhcp/
May 01 12:04:44 maverick-up dhclient[598]:
May 01 12:04:44 maverick-up ifup[448]: Cannot find device "eth0"
May 01 12:04:44 maverick-up ifup[448]: Failed to get interface index: No such device
May 01 12:04:44 maverick-up ifup[448]: If you think you have received this message due to a bug rather
May 01 12:04:44 maverick-up ifup[448]: than a configuration issue please read the section on submitting
May 01 12:04:44 maverick-up ifup[448]: bugs on either our web page at www.isc.org or in the README file
May 01 12:04:44 maverick-up ifup[448]: before submitting a bug.  These pages explain the proper
May 01 12:04:44 maverick-up ifup[448]: process and the information we find helpful for debugging..
May 01 12:04:44 maverick-up ifup[448]: exiting.
May 01 12:04:44 maverick-up ifup[448]: ifup: failed to bring up eth0
May 01 12:04:44 maverick-up ifup[448]: Usage: /srv/maverick/software/maverick/bin/network-if-monitor {start|stop} <interface>
May 01 12:04:44 maverick-up ifup[448]: ifup: failed to bring up wcast0
May 01 12:04:44 maverick-up wpa_supplicant[644]: Successfully initialized wpa_supplicant
May 01 12:04:44 maverick-up wpa_supplicant[644]: Could not read interface wlan0 flags: No such device
May 01 12:04:44 maverick-up wpa_supplicant[644]: nl80211: Driver does not support authentication/association or connect commands
May 01 12:04:44 maverick-up wpa_supplicant[644]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
May 01 12:04:44 maverick-up wpa_supplicant[644]: Could not read interface wlan0 flags: No such device
May 01 12:04:44 maverick-up ifup[448]: /etc/network/if-pre-up.d/wpasupplicant: 120: /etc/network/if-pre-up.d/wpasupplicant: cannot create /dev/stderr: No su
May 01 12:04:44 maverick-up ifup[448]: run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
May 01 12:04:44 maverick-up ifup[448]: ifup: failed to bring up wlan0
May 01 12:04:44 maverick-up ifup[448]: ifup: waiting for lock on /run/network/ifstate.wman0
May 01 12:04:52 maverick-up systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
May 01 12:04:52 maverick-up systemd[1]: Failed to start Raise network interfaces.
May 01 12:04:52 maverick-up systemd[1]: networking.service: Unit entered failed state.
May 01 12:04:52 maverick-up systemd[1]: networking.service: Failed with result 'exit-code'.
```

cglusky, 2018-05-08:
I am seeing similar on wlan0 on raspi0.  But the interface works after boot as it gets an IP and I can ssh in.  net-tools has been deprecated for years but it keeps getting used so I assumed everyone just decided to ignore the deprecation.

fnoop, 2018-05-08:
@cglusky What do you mean about net-tools and deprecation?
The wlan0 error above is because the interface has been renamed to wman0.  wlan0 entry is kept so that people can plug in other usb wifi dongles and they will automatically work.  However, the entry produces an error when there isn't such a dongle attached (same with onboard ethernet eth0), need to work out a more elegant way of dealing with this.

cglusky, 2018-05-08:
Just that net-tools and ifconfig has hung in there so long after the announcement.  Looks like next year is the ten year anniversary of the Debian devlist post.  Should Maverick be using ifup/ifdown?  I do not know enough to say.  Might be a good time to address it if it is an issue.

https://serverfault.com/questions/458628/should-i-quit-using-ifconfig

https://serverfault.com/questions/633087/where-is-the-statement-of-deprecation-of-ifconfig-on-linux

fnoop, 2018-05-08:
Where are we using/depending on ifconfig?
Yes, ifup is integral part of the standard (non-NetworkManager) networking.

cglusky, 2018-05-08:
I thought ifupdown was part of network-tools deprecation since there is/was a dependency?

Stuff like this confuses mere mortals like me: https://blueprints.launchpad.net/ubuntu/+spec/foundations-aa-migrating-to-netplan

cglusky, 2018-05-08:
It seems there is a lot of confusion around this issue so do not think it's critical at this moment.  The Debian docs on networking mark ifupdown as "legacy" but do not offer real alternatives from what I can tell.
https://www.debian.org/doc/manuals/debian-reference/ch05

Either way, really emphasizes how awesome it is that Maverick can abstract away this churn from users like me.

fnoop, 2018-05-31:
Remove automated merging of eth0 and wlan0 default interfaces.  These don't always exist and therefore cause networking service to fail.
Instead, create them in the sample-nodes manifests so they are the default on platforms that do have them, for example include eth0 definition on raspberry but not on up core.
This is a fairly major change and will remove some fallback defaults, but in general more correct.

closed 2018-05-31 by fnoop