goodrobots/maverick #773: RPi 3B+ Support

cglusky, 2018-05-23:
Initial boot with 1.61

Boots fast.  Login and get this...

> WiFi is disabled because the country is not set.
> Use raspi-config to set the country before use.

Could add it to wpa_supplicant.conf

cglusky, 2018-05-23:
Setting wifi country via raspi-config fixes it.  Not sure what file that config is touching.  Need to test 

`country=US`

in wpa_supplicant

cglusky, 2018-05-23:
Getting 

> Hardware platform not recognized

Not surprising.

cglusky, 2018-05-23:
So far everything going well with 3B+ besides the WiFi country not being set complaint.

Seeing errors in configure with ...

Unable to locate package libsodium-dev

So wifibc install fails due lack of dependencies.

No big deal right now since don't plan to use wifibc on this install.  But it was working with Pi0.

cglusky, 2018-05-23:
Another night of linux hell.

Can get everything to work - including some nice video from visiond into QGC - but the mavlink telemetry connection refuses to work.  Tried everything I could think of.

Perhaps the configure portion did not open a UDP port or something.  I have no idea.

Changed the hostname to -minimal to try to reduce variables and now it won't even pick up an IP.

Maybe another day.

cglusky, 2018-05-23:
OH, forgot to mention that the raspi-config command to add/change wifi country just adds the line to top of wpa_supplicant so that just needs to be fixed in the docs for raspi quick wifi setup using wpa_supplicant.conf on the boot drive.

fnoop, 2018-05-23:
Hmm so the wifi is a recent change in raspbian:
```
2018-03-13:
  * Raspberry Pi 3 B+ support
  * WiFi is disabled until wireless regulatory domain is set (Pi 3 B+ only)
    - The domain can be done through 'Raspberry Pi Configuration' (rc_gui),
      'raspi-config' or by setting 'country=' to an appropriate ISO 3166
      alpha2 country code in /etc/wpa_supplicant/wpa_supplicant.conf.
  * Default wireless regulatory domain is now unset
```
Will have to find the best way to disable this behaviour and set to some sensible default.  wpa_supplicant.conf doesn't always have to be in place (and isn't unless it's deliberately added to the boot partition), so that's not necessarily the best solution.

fnoop, 2018-05-23:
Oh, simple enough to disable, it's implemented through a wifi-country systemd manifest.  Simply disable the manifest at boot.

fnoop, 2018-05-23:
Discuss on gitter, could be a number of things.  Not relevant to 3b+ support.

cglusky, 2018-05-24:
So something strange going on with 3B+ and ttyAMA0

I am assuming bluetooth is not being disabled from ttyAMA0 as it is not available for telemetry use.

If I change mavlink-route-fc.conf to use ttyS0 it connects immediately.

And thanks for the log output help as that is what helped me track this down!

With AMA0 It was spitting out "Unidentified autopilot..." so I knew it was at least seeing the port.

With S0 it connected immediately.

cglusky, 2018-05-24:
And the libsodium-dev errors did not come up this time either.  I still think I have transient issues with my wifi that causes those transient errors.  Makes me wonder if there is a way to make those github and apt get downloads more defensive if the connection fails?

cglusky, 2018-05-24:
My temp fix to get AMA0 back for GPIO pin use was to edit config.txt and enable the overlay...

`dtoverlay=pi3-disable-bt`

I am assuming that maverick is already killing the associated bluetooth services

fnoop, 2018-05-28:
@cglusky maverick already has `dtoverlay=pi3-disable-bt` set

fnoop, 2018-05-28:
Ah, it's only set for specific models:
```
    if ($raspberry_model == "3 Model B" or $raspberry_model == "Zero W") and $serialoverride == true {
        confline { "rpi3-uartoverlay":
            file        => "/boot/config.txt",
            line        => "dtoverlay=pi3-disable-bt",
        }
```
I'll add 3b+

closed 2018-05-28 by fnoop