goodrobots/maverick #777: vision_seek service can not be started

lightzl, 2018-06-02:
I have a seek thermal pro and a raspberry pi 0w. I tried to make them work together, following the installation instruction. 
I downloaded Raspberry Pi Lite (Pi Zero/W) - 1.1.6.
Wrote it to the SD card.
Configured the WiFi.
Plugged it and powered on.
Connected by SSH and log in.
Run the commands "maverick self-update" and "maverick configure" , then reboot.
When I check the running service with "maverick status", all vision services are stopped and disabled, including vision_seek. So I try "maverick start vision_seek", and it shows start success. But when I check the service again, the service is still stopped. I try to start visiond service. It,s also not work. But when I tried to start webvision service. It worked. lsusb shows the device is identified to 0011.

What should I do?

fnoop, 2018-06-02:
Hi, I've not tried the 'pro', but it should work.  First thing is to get vision_seek running.  Have you edited ~/config/vision/vision_seek.conf and changed CAMTYPE to seekpro? (That should be in the docs!)
If you like, head over to the gitter channel for realtime help.

lightzl, 2018-06-03:
Thank you for you help.  The changing camtype to seekpro is working. I got the seek.avi file. But very short. Vision_seek service is seems like working only a few seconds. Then stop. Similar to this issue[https://github.com/goodrobots/maverick/issues/776](url). The difference is my device is pi 0w. My image seems to be broken when I replug seek pro hot. The pi 0w restart repeatedly. I will try later. 
How can I get to the gitter? Could you give me a link?

fnoop, 2018-06-15:
@lightzl How are you getting on with this?  Gitter is here: https://gitter.im/goodrobots/maverick

alfax21, 2018-09-14:
`2018-09-11 22:57:39,019 - INFO - Starting maverick-visiond
2018-09-11 22:57:39,032 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 22:57:39,040 - CRITICAL - Error detecting camera video device
2018-09-11 22:58:17,902 - INFO - Starting maverick-visiond
2018-09-11 22:58:17,916 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 22:58:17,935 - CRITICAL - Error detecting camera video device
2018-09-11 22:58:56,922 - INFO - Starting maverick-visiond
2018-09-11 22:58:56,941 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 22:58:56,950 - CRITICAL - Error detecting camera video device
2018-09-11 22:59:34,178 - INFO - Starting maverick-visiond
2018-09-11 22:59:34,213 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 22:59:34,241 - CRITICAL - Error detecting camera video device
2018-09-11 23:00:11,255 - INFO - Starting maverick-visiond
2018-09-11 23:00:11,313 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:00:11,326 - CRITICAL - Error detecting camera video device
2018-09-11 23:00:48,031 - INFO - Starting maverick-visiond
2018-09-11 23:00:48,046 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:00:48,055 - CRITICAL - Error detecting camera video device
2018-09-11 23:01:24,667 - INFO - Starting maverick-visiond
2018-09-11 23:01:24,689 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:01:24,723 - CRITICAL - Error detecting camera video device
2018-09-11 23:02:01,512 - INFO - Starting maverick-visiond
2018-09-11 23:02:01,551 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:02:01,559 - CRITICAL - Error detecting camera video device
2018-09-11 23:02:38,734 - INFO - Starting maverick-visiond
2018-09-11 23:02:38,761 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:02:38,770 - CRITICAL - Error detecting camera video device
2018-09-11 23:03:16,438 - INFO - Starting maverick-visiond
2018-09-11 23:03:16,473 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:03:16,493 - CRITICAL - Error detecting camera video device
2018-09-11 23:03:53,002 - INFO - Starting maverick-visiond
2018-09-11 23:03:53,026 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:03:53,046 - CRITICAL - Error detecting camera video device
2018-09-11 23:04:29,991 - INFO - Starting maverick-visiond
2018-09-11 23:04:30,005 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:04:30,014 - CRITICAL - Error detecting camera video device
2018-09-11 23:05:07,512 - INFO - Starting maverick-visiond
2018-09-11 23:05:07,548 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:05:07,560 - CRITICAL - Error detecting camera video device
2018-09-11 23:05:43,993 - INFO - Starting maverick-visiond
2018-09-11 23:05:44,045 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:05:44,053 - CRITICAL - Error detecting camera video device
2018-09-11 23:06:20,853 - INFO - Starting maverick-visiond
2018-09-11 23:06:20,865 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:06:20,873 - CRITICAL - Error detecting camera video device
2018-09-11 23:06:57,966 - INFO - Starting maverick-visiond
2018-09-11 23:06:57,988 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:06:58,006 - CRITICAL - Error detecting camera video device
2018-09-11 23:07:35,445 - INFO - Starting maverick-visiond
2018-09-11 23:07:35,481 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:07:35,527 - CRITICAL - Error detecting camera video device
2018-09-11 23:08:14,242 - INFO - Starting maverick-visiond
2018-09-11 23:08:14,275 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:08:14,309 - CRITICAL - Error detecting camera video device
2018-09-11 23:08:53,402 - INFO - Starting maverick-visiond
2018-09-11 23:08:53,426 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:08:53,447 - CRITICAL - Error detecting camera video device
2018-09-11 23:09:33,045 - INFO - Starting maverick-visiond
2018-09-11 23:09:33,064 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:09:33,075 - CRITICAL - Error detecting camera video device
2018-09-11 23:10:12,306 - INFO - Starting maverick-visiond
2018-09-11 23:10:12,318 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:10:12,326 - CRITICAL - Error detecting camera video device
2018-09-11 23:10:49,809 - INFO - Starting maverick-visiond
2018-09-11 23:10:49,840 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:10:49,849 - CRITICAL - Error detecting camera video device
2018-09-11 23:11:28,656 - INFO - Starting maverick-visiond
2018-09-11 23:11:28,688 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-11 23:11:28,733 - CRITICAL - Error detecting camera video device
2018-09-11 23:12:08,260 - INFO - Starting maverick-visiond
[... 904 lines cut ...]
2018-09-14 10:11:56,669 - CRITICAL - Error detecting camera video device
[... cut ...]
2018-09-14 10:15:58,407 - CRITICAL - Error detecting camera video device
2018-09-14 10:16:40,815 - INFO - Starting maverick-visiond
2018-09-14 10:16:40,875 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-14 10:16:40,905 - CRITICAL - Error detecting camera video device
2018-09-14 10:17:23,436 - INFO - Starting maverick-visiond
2018-09-14 10:17:23,481 - INFO - Config read from /srv/maverick/config/vision/maverick-visiond.conf
2018-09-14 10:17:23,491 - CRITICAL - Error detecting camera video device`