goodrobots/maverick #781: eth1 interface not avilable after desktop enable - RPI / streach

Mocart96, 2018-06-20:
For a last couplle of day I was trying to enable at but bt hilnk lte modem. Modem is recognized   at the boot by maverick with eth1 interface but after pixhel/sheld enable it dissapear. Finally I found solution by adding eth1 inteface as a command to /etc/rc.local file and it works now,
could you fix this issuse with net realises to enable hilink lte modems,

fnoop, 2018-06-20:
Hi, what command have you added to /etc/rc.local to get it to work?
I've never used lte modems, but if they present as ethernet and are setup through dhcp it should work out of the box.  There's no automatic definition for eth1 however, so it may be that you have to add one in localconf.json

Mocart96, 2018-06-20:
Hi, I did add "sudo dhclient eth1" only. After this interface is ready. Will attach screen when I back home.

Mocart96, 2018-06-20:
@fnoop Below screens how it look like just after boot. By editing rc.local I able to bring back eth1 to interface and Everything works fine.

1. at the first boot eth1 is available:

[image]

2. after desktop enable eth1 is missing 

[image]