goodrobots/maverick #788: Bootstrap fails on digitalocean droplet with locale errors

fnoop, 2018-09-11:
Bootstrap should work out of the box with a digitalocean droplet

SamuelDudley, 2018-09-11:
Initial bootstrap
[maverick.bootstrap.2018-09-11-11-57.log](https://github.com/goodrobots/maverick/files/2371021/maverick.bootstrap.2018-09-11-11-57.log)
first run of the dev configure (note errors)
[maverick.dev.2018-09-11-12-34.log](https://github.com/goodrobots/maverick/files/2371023/maverick.dev.2018-09-11-12-34.log)

fnoop, 2018-09-18:
```
Notice: Scope(Oncevcsrepo[git-maverick]): oncevcsrepo: /srv/maverick/software/maverick git repo doesn't exist locally, cloning may take a while..
Notice: Scope(Install_python_module[pip-pyric]): Installing pip: PyRIC
Notice: Scope(Install_python_module[pip-netifaces]): Installing pip: netifaces
Notice: Scope(Install_python_module[pip-future]): Installing pip: future
Notice: Compiled catalog for goodrobots in environment bootstrap in 1.00 seconds
Notice: /Stage[bootstrap]/Base::Locale/Package[locales-all]/ensure: created
Notice: /Stage[bootstrap]/Locales/File[/etc/locale.gen]/content: content changed '{md5}6b8eb47d184249a7f976e5877ac4cc5a' to '{md5}de2f35f6889f95ce9f4faa61177858c1'
Notice: /Stage[bootstrap]/Locales/Exec[locale-gen]: Triggered 'refresh' from 1 event
Notice: /Stage[bootstrap]/Locales/File[/etc/default/locale]/content: content changed '{md5}bb0cbb13b50e21e5709822e3b69cf4c6' to '{md5}7ff8fd624821db587d69f8c2e2e37698'
Notice: /Stage[bootstrap]/Locales/Exec[update-locale]/returns: *** update-locale: Error: invalid locale settings:  LANG="en_GB.UTF-8"
Error: /Stage[bootstrap]/Locales/Exec[update-locale]: Failed to call refresh: '/usr/sbin/update-locale' returned 255 instead of one of [0]
Error: /Stage[bootstrap]/Locales/Exec[update-locale]: '/usr/sbin/update-locale' returned 255 instead of one of [0]
Notice: /Stage[bootstrap]/Base::Locale/Exec[tz-systemd]/returns: executed successfully
Notice: /Stage[bootstrap]/Base::Locale/File[/etc/profile.d/01-TZ.sh]/ensure: defined content as '{md5}bd2120a9d614cfcbcff44e12341db4b9'
Notice: /Stage[bootstrap]/Base::Packages/Package[iotop]/ensure: created
Notice: /Stage[bootstrap]/Base::Packages/Package[v4l-utils]/ensure: created
Notice: /Stage[bootstrap]/Base::Packages/Package[cmake]/ensure: created
```

closed 2019-05-02 by fnoop