goodrobots/maverick #795: One source of mavlink messages

SamuelDudley, 2018-10-23:
Many of the projects that Maverick sets up use mavlink for communication but a single source is not used to define and generate the messages. It would be great if for example MAVProxy used the same mavlink messages definitions as the ArduPilot code base. That way when adding messages to the ArduPilot codebase they are supported by the GCS software.

fnoop, 2019-03-27:
MAVProxy is installed through pip and has pymavlink as a dependency, is python and pymavlink contains it's own mavlink messages.  Ardupilot is c++ and has it's own mavlink submodule.  I don't see how Maverick could solve this, or how it could be in scope for Maverick?

Did you have an idea how this might be solved?

SamuelDudley, 2019-03-27:
My usual development practice is to git clone both Ardupilot and MAVProxy and then install all of the requirements for MAVProxy via pip (but not pymavlink)

Mavlink is a submodule of Ardupilot and pymavlink is a submodule of mavlink, so by git cloning Ardupilot and its submodules you end up with a copy of pymavlink on the system (but the python interpreter is not aware of its location)

I then add the pymavlink submodule folder to the $PYTHONPATH which satisfies the pymavlink requirement for MAVProxy. MAVProxy now functions as expected.

Now when I make changes in my mavlink message .xml definition(s) to support a feature in Ardupilot they are also automatically picked up by MAVProxy

The biggest change is the requirement to git clone MAVProxy rather than pip install it....