goodrobots/maverick #796: Mavproxy Heartbeat Re: Pixhawk and Pi Connection

eanmikale, 2018-10-30:
fnoop, 
Hey, I am having a bit of trouble getting the heartbeat from the pixhawk using maverick. We have correct connections from telemetry 2 to the GPIO pins. These are also the steps we have taken as you can see below...

1) Change Local.config file for Maverick, making sure to specify mavlink input at “serial1”, and mavlink_proxy as “mavproxy”.  
2) Set Pixhawk Parameters inside Qground Control
	SERIAL2_PROTOCOL = 1 (the default) to enable MAVLink on the serial port. 
SERIAL2_BAUD = 921 so the Pixhawk can communicate with the RPi at 921600 baud. 
LOG_BACKEND_TYPE = 3 if you are using APSync to stream the dataflash log files to the RPi 
3) Disable Serial Port on Raspberry Pi and reboot
4) Enable UART connection to 1 by edit /boot/config.txt
5) Disable bluetooth dtoverlay=pi3-disable-bt and dtoverlay=pi3-miniuart-bt
	and run sudo systemctl disable hciuart
5) Change /boot/cmdline.txt by changing first TTYAMA0/TTYS0 to “serial1,115200”
6) Edit file /lib/systemd/system/hciuart.service to specify serial 1 
7) Gain super user permissions via sudo -s
8) Test connection via command = mavproxy.py --master=/dev/serial1 --baudrate 57600 --aircraft MyCopter

AND HERE IS THE OUTPUT

root@maverick-raspberry:/dev# mavproxy.py --master=/dev/serial1 --baudrate 115200 --aircraft MyCopter
Connect /dev/serial1 source_system=255
Failed to load module: No module named adsb. Use 'set moddebug 3' in the MAVProxy console to enable traceback
no script MyCopter/mavinit.scr
Log Directory: MyCopter/logs/2018-10-30/flight3
Telemetry log: MyCopter/logs/2018-10-30/flight3/flight.tlog
MAV> Waiting for heartbeat from /dev/serial1

ANY HELP IS APPRECIATED! Thanks in advance.

fnoop, 2018-10-30:
Hi @eanmikale, do you have a particular reason for all those steps and messing around with the uarts?  Maverick sets everything up for you and has the correct overlays and port settings in place for the raspberry, to be reliable with the hardware uart active on /dev/ttyAMA0.
Also you seem to be setting the Fc at 92100 baud, cmdline.txt to 115200 baud and mavproxy to 57600, and you have some conflicting actions in your steps (for example you disable hciuart and then configure it 2 steps later).
Unless you have a particularly good reason for doing all this stuff, just let Maverick do it's thing and use ttyAMA0 - it's all setup for you and works well.  Also unless you particularly need something in mavproxy, I would recommend the default mavlink-router - it's much more efficient on lower-power board like the raspberry.

eanmikale, 2018-10-30:
Fnoop,  Thank you for your time. I will reconfigure to overwrite current settings to defaults tonight. Is there any reason that the file /lib/systemd/system/hicuart.service specifies serial1? Is this accomplished in the local.config file? The raspberry pi documentation, as well as many other developers have used the serial1 or serial0 ports as opposed to the ttyAMA0 ports. However, we desire to utilize maverick exclusively, and prefer to understand how the maverick platform is specifically calibrated to interact with the Pixhawk. We look forward to your feedback. More soon.  Regards,  Ean Mikale, J.D. Chief Innovation Officer Infinite 8 Institute, L3C ean@infinite8institute.com www.infinite8institute.com U.S. Federal Contractor ---- On Tue, 30 Oct 2018 17:36:37 -0500 Fnoop <notifications@github.com> wrote ---- Hi @eanmikale, do you have a particular reason for all those steps and messing around with the uarts? Maverick sets everything up for you and has the correct overlays and port settings in place for the raspberry, to be reliable with the hardware uart active on /dev/ttyAMA0. Also you seem to be setting the Fc at 92100 baud, cmdline.txt to 115200 baud and mavproxy to 57600, and you have some conflicting actions in your steps (for example you disable hciuart and then configure it 2 steps later). Unless you have a particularly good reason for doing all this stuff, just let Maverick do it's thing and use ttyAMA0 - it's all setup for you and works well. Also unless you particularly need something in mavproxy, I would recommend the default mavlink-router - it's much more efficient on lower-power board like the raspberry.

eanmikale, 2018-10-31:
Fnoop, 
Issue fixed. After making your recommended changes, I switched from powering the Pi via 5v to USB. Upon doing this, the port ttyAMC0 appeared under /dev/. This was the correct port. Thank you for your time.

closed 2018-10-31 by eanmikale

fnoop, 2018-11-02:
Hi @eanmikale, so serial0 and serial1 not real devices - they're aliases to the actual device.  If you look in /dev, you'll see something like:  
```
[dev] [mav@maverick-raspberry ~]$ ls -l /dev/serial0
lrwxrwxrwx 1 root root 7 Nov  1 19:17 /dev/serial0 -> ttyAMA0
[dev] [mav@maverick-raspberry ~]$ ls -l /dev/serial1
lrwxrwxrwx 1 root root 5 Nov  1 19:17 /dev/serial1 -> ttyS0
```
The device mappings are confusing enough, so most people find it easier to ignore these aliases and just use the devices directly.

Next, maverick disables bluetooth altogether by using the pi3-disable-bt overlay.  This is because bluetooth is not really used in robotics (I've never heard of anyone using it in a UAV), so it's better to remove it to save further confusion and to remove an unnecessary source of EMI.  With this we also disable hciuart service.  I see the raspberry docs recommend editing the systemd manifest (/lib/systemd/system/hciuart.service) if you are switching the UARTs around, but this is really bad advice - you should never edit root-owned, package installed files, and this will get overwritten by subsequent package updates.  Otherwise this is a really good source of info:
https://www.raspberrypi.org/documentation/configuration/uart.md

So what we're left with is /dev/ttyAMA0 mapped to the real hardware UART, pointing to the GPIO UART pins that we need to communicate with a flight controller.  This is the most reliable, simplest setup for serial communication on the raspberry.  All you need to do is set the serial port on the flight controller to Mavlink (1 or 2) protocol, baud rate 115200 (you can use other baud rates but this is the most reliable to start with), and connect it to a freshly installed Maverick - it should just work with no further setup.

Hope this helps :)

eanmikale, 2018-11-02:
Fnoop,
Thank you for your time. This brings a much deeper clarity. Mav is a phenomenal open-source project the industry needed desperately to move innovation forward at a more rapid and globally competitive pace. Here's to commercialization.

Cheers!

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Fri, 02 Nov 2018 02:31:53 -0500 notifications@github.com wrote ----

Hi @eanmikale, so serial0 and serial1 not real devices - they're aliases to the actual device. If you look in /dev, you'll see something like:

[dev] [mav@maverick-raspberry ~]$ ls -l /dev/serial0
lrwxrwxrwx 1 root root 7 Nov  1 19:17 /dev/serial0 -> ttyAMA0
[dev] [mav@maverick-raspberry ~]$ ls -l /dev/serial1
lrwxrwxrwx 1 root root 5 Nov  1 19:17 /dev/serial1 -> ttyS0

The device mappings are confusing enough, so most people find it easier to ignore these aliases and just use the devices directly.

Next, maverick disables bluetooth altogether by using the pi3-disable-bt overlay. This is because bluetooth is not really used in robotics (I've never heard of anyone using it in a UAV), so it's better to remove it to save further confusion and to remove an unnecessary source of EMI. With this we also disable hciuart service. I see the raspberry docs recommend editing the systemd manifest (/lib/systemd/system/hciuart.service) if you are switching the UARTs around, but this is really bad advice - you should never edit root-owned, package installed files, and this will get overwritten by subsequent package updates. Otherwise this is a really good source of info:
https://www.raspberrypi.org/documentation/configuration/uart.md

So what we're left with is /dev/ttyAMA0 mapped to the real hardware UART, pointing to the GPIO UART pins that we need to communicate with a flight controller. This is the most reliable, simplest setup for serial communication on the raspberry. All you need to do is set the serial port on the flight controller to Mavlink (1 or 2) protocol, baud rate 115200 (you can use other baud rates but this is the most reliable to start with), and connect it to a freshly installed Maverick - it should just work with no further setup.

Hope this helps :)