goodrobots/maverick #797: Document custom maverick module

fnoop, 2018-11-07:
eg. how to create a custom module to add a second SITL instance

fnoop, 2018-11-07:
And is there somewhere outside of ~/software/maverick that a custom module can be placed?  ~/software/maverick is goodrobots git repo, either a custom module directory can be .gitignored, or probably better a separate area in ~/code should be setup.

fnoop, 2018-11-12:
Add two locations for custom modules:
 - ~/software/maverick/manifests/custom-modules
 - ~/code/maverick/custom-modules

fnoop, 2018-11-12:
Interface stats added to stats module.

fnoop, 2018-11-12:
Oops, wrong repo

fnoop, 2018-11-12:
Added support for the new module locations.  Just need to document.

rgreenosborne, 2018-12-13:
https://github.com/rgreenosborne/maverick/blob/master/docs/next/modules/custom.md
      
@fnoop Am I doing this right?  do you have a code block for an actual second SITL module? Git is so damned intimidating.

closed 2020-04-29 by fnoop