goodrobots/maverick #806: Grafana fails to install

SamuelDudley, 2019-01-13:
Running `maverick configure` graphana fails to install
Cause: `latest` is being used in the download url
Fix: change `$grafana_version = latest,` to `$grafana_version = '5.1.4',`

fnoop, 2019-03-23:
The old thirdparty source for raspberry packages has stopped, but grafana is producing arm packages now, so switch all sources to grafana packages.
They provide an apt repo now:
http://docs.grafana.org/installation/debian/

Only armv7 unfortunately so we'll have to drop support for rpi zero.

I think we'll stop using grafana shortly anyway as it doesn't support integration - it demands to be used as a complete interface.

fnoop, 2019-03-23:
Latest grafana 6.0.2 doesn't work with the existing config, but latest 5.0 (5.4.3) version does so fixing that.

closed 2019-03-23 by fnoop