goodrobots/maverick #808: Opencv fails to build

SamuelDudley, 2019-01-13:
__Issue:__ running `maverick configure` fails during the opencv build. All vision code subsequently fails.
See attached log files:
https://files.gitter.im/goodrobots/dev/Lkew/maverick.dev.2019-01-13-03-07.log
https://files.gitter.im/goodrobots/dev/LCv0/CMakeOutput.log__1_.tar.gz

__Env:__ current maverick master on a Pi3b+

SamuelDudley, 2019-01-19:
Some more info on this.... it was `opencv_dldt` failing in `opencv_dldt-prepbuild`

fnoop, 2019-01-19:
Ah that's openvino: https://github.com/goodrobots/maverick/issues/791

This has been resolved by the upstream project now

fnoop, 2019-01-19:
Wow my issue really opened a can of worms!  They haven't tagged the R5 release in github so I'll just change the reference to master for now.

fnoop, 2019-03-22:
Although it should be fixed and compile, openvino is now turned off by default anyway, as it is of limited use without specific hardware requirements.

closed 2019-03-22 by fnoop