goodrobots/maverick #815: Add support for intel realsense sdk 2

fnoop, 2019-03-16:


fnoop, 2019-03-16:
This is the legacy branch:
https://github.com/IntelRealSense/librealsense/tree/legacy

This is the new branch (sdk2):
https://github.com/IntelRealSense/librealsense/tree/master

alchemyconsulting, 2019-03-22:
So realsense currently exists in:
 maverick_hardware::peripheral::realsense
and
 maverick_vision::realsense

Originally librealsense and realsense-sdk (and samples) were separate projects and installs.  Now librealsense is the sdk as well, so we should get rid of one of them.  Which one?

fnoop, 2019-03-22:
So realsense currently exists in:
maverick_hardware::peripheral::realsense
and
maverick_vision::realsense

Originally librealsense and realsense-sdk (and samples) were separate projects and installs. Now librealsense is the sdk as well, so we should get rid of one of them. Which one?

Keep maverick_hardware, ditch maverick_vision

fnoop, 2019-03-22:
Python wrapper:
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python

closed 2019-03-23 by fnoop