goodrobots/maverick #823: Add support for Jetson Nano

fnoop, 2019-03-28:


fnoop, 2019-03-28:
Missing components and module _ctypes fails to build when building custom python:
```
Python build finished successfully!
The necessary bits to build these optional modules were not found:
_curses               _curses_panel         _dbm
_gdbm                 _lzma                 _tkinter
readline
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py, have been
built by the Makefile instead, as configured by the Setup files:
_abc                  atexit                pwd
time

Failed to build these modules:
_ctypes
```

fnoop, 2019-03-28:
```
Error: '/usr/bin/make install >>/srv/maverick/var/log/build/jtx1inst.build.out 2>&1' returned 2 instead of one of [0]
Error: /Stage[main]/Maverick_hardware::Tegra/Exec[jtx1inst-install]/returns: change from 'notrun' to ['0'] failed: '/usr/bin/make install >>/srv/maverick/var/log/build/jtx1inst.build.out 2>&1' returned 2 instead of one of [0]
Notice: /Stage[main]/Maverick_hardware::Tegra/File[/srv/maverick/software/jtx1inst/bin]: Dependency Exec[jtx1inst-install] has failures: true
Warning: /Stage[main]/Maverick_hardware::Tegra/File[/srv/maverick/software/jtx1inst/bin]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_hardware::Tegra/Exec[cp-jtx1inst-bin]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_hardware::Tegra/File[/srv/maverick/var/build/.install_flag_jtx1inst]: Skipping because of failed dependencies
```

fnoop, 2019-03-28:
```
Error: Could not set 'directory' on ensure: Could not find user openldap (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_security/manifests/ldap_server.pp, line: 19)
Error: Could not set 'directory' on ensure: Could not find user openldap (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_security/manifests/ldap_server.pp, line: 19)
Wrapped exception:
Could not find user openldap
Error: /Stage[main]/Maverick_security::Ldap_server/File[/etc/ldap/ssl]/ensure: change from 'absent' to 'directory' failed: Could not set 'directory' on ensure: Could not find user openldap (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_security/manifests/ldap_server.pp, line: 19)
Notice: /Stage[main]/Maverick_security::Ldap_server/Exec[create-ldapssl-key]: Dependency File[/etc/ldap/ssl] has failures: true
Warning: /Stage[main]/Maverick_security::Ldap_server/Exec[create-ldapssl-key]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/Exec[create-ldapssl-csr]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/File[/etc/ldap/ssl/slapd-ldapssl.ext]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/Exec[create-ldapssl-cert]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/File[/etc/ldap/ssl/slapd-ldapssl.key]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/File[/etc/ldap/ssl/slapd-ldapssl.csr]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_security::Ldap_server/File[/etc/ldap/ssl/slapd-ldapssl.crt]: Skipping because of failed dependencies
```

This might be a general problem, not specific to jetson nano

fnoop, 2019-03-28:
```
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: 	C compiler:		gcc
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: 	C++ compiler:		g++ -std=gnu++11
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:   GEN      include/mavlink/ardupilotmega/mavlink.h
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: Traceback (most recent call last):
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:   File "./modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:     from pymavlink.generator import mavgen
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:   File "/srv/maverick/var/build/mavlink-router/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns:     from future import standard_library
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: ModuleNotFoundError: No module named 'future'
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: Makefile:1609: recipe for target 'include/mavlink/ardupilotmega/mavlink.h' failed
Notice: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: make: *** [include/mavlink/ardupilotmega/mavlink.h] Error 1
Error: '/srv/maverick/var/build/mavlink-router/autogen.sh && CFLAGS='-g -O2' /srv/maverick/var/build/mavlink-router/configure --prefix=/srv/maverick/software/mavlink-router --disable-systemd && /usr/bin/make -j2 && make install >/srv/maverick/var/log/build/mavlink-router.build.out 2>&1' returned 2 instead of one of [0]
Error: /Stage[main]/Maverick_mavlink/Exec[mavlink-router-build]/returns: change from 'notrun' to ['0'] failed: '/srv/maverick/var/build/mavlink-router/autogen.sh && CFLAGS='-g -O2' /srv/maverick/var/build/mavlink-router/configure --prefix=/srv/maverick/software/mavlink-router --disable-systemd && /usr/bin/make -j2 && make install >/srv/maverick/var/log/build/mavlink-router.build.out 2>&1' returned 2 instead of one of [0]
Notice: /Stage[main]/Maverick_mavlink/File[/srv/maverick/var/build/.install_flag_mavlink-router]: Dependency Exec[mavlink-router-build] has failures: true
Warning: /Stage[main]/Maverick_mavlink/File[/srv/maverick/var/build/.install_flag_mavlink-router]: Skipping because of failed dependencies
```

fnoop, 2019-03-28:
OpenCV not building cuda modules:
`--     Unavailable:                 cnn_3dobj cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev cvv java js matlab ovis sfm viz`

fnoop, 2019-03-28:
https://docs.opencv.org/trunk/d6/d15/tutorial_building_tegra_cuda.html
https://jkjung-avt.github.io/opencv3-on-tx2/
https://www.jetsonhacks.com/2018/05/28/build-opencv-3-4-with-cuda-on-nvidia-jetson-tx2/

fnoop, 2019-03-30:
Tensorflow not installing:
```
Notice: Scope(Install_python_module[tensorflow-pip]): Installing pip: tensorflow
Notice: Compiled catalog for maverick-jetsonnano.lan in environment dev in 8.47 seconds
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: Collecting tensorflow
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns:   Could not find a version that satisfies the requirement tensorflow (from versions: )
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: No matching distribution found for tensorflow
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: Collecting tensorflow
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns:   Could not find a version that satisfies the requirement tensorflow (from versions: )
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: No matching distribution found for tensorflow
Error: 'pip3 wheel --help > /dev/null 2>&1 && { pip3 show wheel > /dev/null 2>&1 || wheel_support_flag='--no-binary :all:'; } ; { pip3 --log /tmp/pip.log install $wheel_support_flag   --disable-pip-version-check  tensorflow || pip3 --log /tmp/pip.log install   --disable-pip-version-check  tensorflow ;}' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: change from 'notrun' to ['0'] failed: 'pip3 wheel --help > /dev/null 2>&1 && { pip3 show wheel > /dev/null 2>&1 || wheel_support_flag='--no-binary :all:'; } ; { pip3 --log /tmp/pip.log install $wheel_support_flag   --disable-pip-version-check  tensorflow || pip3 --log /tmp/pip.log install   --disable-pip-version-check  tensorflow ;}' returned 1 instead of one of [0]
```

fnoop, 2019-03-30:
Grafana:
 - Power consumption not working
 - Add GPU usage

eanmikale, 2019-05-05:
@fnoop - Any updates on this build? Thank you.

fnoop, 2019-05-15:
jtx1inst doesn't build: https://github.com/cezs/jtx1inst/issues/9

fnoop, 2019-05-15:
@eanmikale working on it!  should have something to test soon-ish

eanmikale, 2019-05-15:
@fnoop - We've had some weird experiences with the Nano. We've seen the system act somewhat glitch, stuttering as terminal windows populate and/or disappear. 

We also are selling drone kits. How do we collaborate to include Maverick, with the necessary licensing language? We are selling three systems: 1) a system with the Pixhawk, Pi 3 B/B+, and a touch screen interface, 2) a system including the prior items, in addition to a seek thermal camera, and 3) a system including the prior items (excluding seek thermal), with the Jetson Nano. We haven't gotten a touchscreen to work with the Nano yet. 

Any way we could collaborate to create a workforce of American drone pilots, engineers, and manufacturers, would be appreciated. 

Regards, 

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Wed, 15 May 2019 18:18:19 -0500 notifications@github.com wrote ----

@eanmikale working on it! should have something to test soon-ish

fnoop, 2019-05-16:
Nano build now clean except for tensorflow.  Tensorflow python wheels from nvidia are only for python 3.6 and don't work with our custom 3.7 install.  Options are:
 - Install using system python 3.6
 - Install from source
 - Force install 3.6 wheel in 3.7?

fnoop, 2019-05-16:
Hey @eanmikale.  Haven't started using the nano in anger yet, will do from now as I've got a build going.  Not surprising it glitches - it's a regular desktop ubuntu build with a whole heap of crap added to it - I would call it more a development environment than a production robotic environment.  I'll look to disable/strip a lot of the unnecessary crud in maverick over time.  As a general rule for an embedded system the desktop should be disabled, which should help.

Re: the licensing - all of goodrobots projects are MIT or similar licensed - so called 'permissive' licensing.  This allows you to do whatever you want with the code/project - if you want you can completely rebrand it and sell it for a million dollars if that's what is convenient for you.  We attach absolutely no restrictions whatsoever, we want complete freedom for these projects to help whoever they can help.  The only condition is the copyright must be distributed as per the license:
https://github.com/goodrobots/maverick/blob/stable/LICENSE
There is an open issue around the licensing (https://github.com/goodrobots/maverick/issues/824), but it is for simplication/clarification as the MIT can cause some difficulties.  The basic tenet is that our software is completely free in use and money, and always will be.

We hope in return that at least some of the people or orgs that it helps out, will contribute something back :)

eanmikale, 2019-05-16:
@fnoop - We understand clearly. We look forward to contributing to the ecosystem. More soon. 

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Thu, 16 May 2019 03:21:40 -0500 notifications@github.com wrote ----

Hey @eanmikale. Haven't started using the nano in anger yet, will do from now as I've got a build going. Not surprising it glitches - it's a regular desktop ubuntu build with a whole heap of crap added to it - I would call it more a development environment than a production robotic environment. I'll look to disable/strip a lot of the unnecessary crud in maverick over time. As a general rule for an embedded system the desktop should be disabled, which should help.

Re: the licensing - all of goodrobots projects are MIT or similar licensed - so called 'permissive' licensing. This allows you to do whatever you want with the code/project - if you want you can completely rebrand it and sell it for a million dollars if that's what is convenient for you. We attach absolutely no restrictions whatsoever, we want complete freedom for these projects to help whoever they can help. The only condition is the copyright must be distributed as per the license:
https://github.com/goodrobots/maverick/blob/stable/LICENSE
There is an open issue around the licensing (#824), but it is for simplication/clarification as the MIT can cause some difficulties. The basic tenet is that our software is completely free in use and money, and always will be.

We hope in return that at least some of the people or orgs that it helps out, will contribute something back :)

fnoop, 2019-08-10:
Rebase on new nano SD image based on JetPack 4.2.1

fnoop, 2019-08-16:
Tegra now building cleanly and working well.  Any subsequent problems should be raised as separate issues.

closed 2019-08-16 by fnoop