goodrobots/maverick #824: Licensing, again

fnoop, 2019-03-29:
https://writing.kemitchell.com/2019/03/09/Deprecation-Notice.html

Not suggesting this new license necessarily, but it does seem that Apache 2.0 is a very popular permissive license.  Very large companies like facebook still use MIT but they don't really have to worry too much about being sued..

fnoop, 2019-03-29:
@SamuelDudley @cglusky for discussion?

SamuelDudley, 2019-03-29:
I was reading the post you linked and the associated licence. I'm allergic to layer speak so the blue oak licence is appealing in that regard.

cglusky, 2019-03-29:
I agree. Blue Oak seems really simple and easy to read. Works for me.

SamuelDudley, 2019-04-03:
Based on the above is everyone happy to move licence? I'm for the change.

fnoop, 2019-06-04:
Yeah I like blue oak, I'm for the change.  I'm a little hesitant that it's not a recognised license, probably hasn't been tested or looked at by big companies or lawyers, and most people will never have heard of it.  I suggest that we add a clear note to the license file explaining what it is and that it's a permissive license, and links to one or more of the following pages:
https://blueoakcouncil.org/
https://blueoakcouncil.org/list
https://blueoakcouncil.org/2019/03/06/model.html
https://writing.kemitchell.com/2019/03/09/Deprecation-Notice.html

fnoop, 2019-08-10:
Make decision by 2.0-beta

sskras, 2020-10-04:
@fnoop wrote:

> Yeah I like blue oak, I'm for the change. I'm a little hesitant that it's not a recognised license, probably hasn't been tested or looked at by big companies or lawyers, and most people will never have heard of it. 

Just my .02€: if every project would wait for a license to be popular, there will be no MIT, no GPL, not APL2 present today.

If this project has at least some support, it should go for it. Maybe even go dual license under MIT+BOML for the transition time, if it's very sensitive project.

Cheers!

sskras, 2024-01-29:
BOML just got an OSI approval: https://github.com/spdx/license-list-XML/issues/2352