goodrobots/maverick #826: Python2 deprecation notice - remove python2 altogether?

fnoop, 2019-03-30:
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.

fnoop, 2019-04-25:
This is coming from pip.rb custom fact that extracts list of installed python modules from various pip versions.:
```
    envs = ["maverick", "global", "global3", "sitl", "fc"]
    pippaths = {"maverick" => "/srv/maverick/software/python/bin/pip3", "global" => "pip", "global3" => "pip3", "sitl" => "/srv/maverick/.virtualenvs/sitl/bin/pip", "fc" => "/srv/maverick/.virtualenvs/fc/bin/pip"}
```

The 'global' pip refers to /usr/bin/pip on most systems which is the old pip2.  This is producing the deprecation error when run under specific circumstances.

In addition, ~/software/maverick/bin/maverick does a check and install of python during the initial setup.  This should be disabled if we're now doing all python with our custom python install.

fnoop, 2019-04-25:
Still need to keep support for pip2 in install_python_module and hence the custom fact, otherwise users won't be able to explicitly manage python2 modules, which is a valid use case.

fnoop, 2019-04-25:
If we completely get rid of python2 then none of the other custom facts run.  Need to keep it installed for basic functionality, before our custom python is installed.

fnoop, 2020-01-31:
Reopening, I think the time has come to eradicate python2

fnoop, 2020-01-31:
@SamuelDudley - do you think we're safe now to completely eradicate python2?
I'm switching more and more things over to python3 but having issues because there are still quite a lot of scripts that have `#!/usr/bin/env python` as their hashbang.
/srv/maverick/software/python/bin is already priority in the system path so it should be called first, but there is no 'python' command/link by default installed with python3.  So I can just create a python symlink pointing to python3, and that should be enough for most scripts to then pick it up, in favour of /usr/bin/python.

Separately, I can work on actually force removing the python2 packages.  There are *so* many packages that depend on them though, so might not be possible.

SamuelDudley, 2020-02-02:
Sorry, saw this in my feed then missed it! I'm okay with the symlink plan but we need to be sure that we aren't breaking existing distribution packages. 
I guess a good trial by fire is to try an bootstrap maverick with the symlink in place?

fnoop, 2020-02-02:
Can't remove python2:
```
[dev] [mav@maverick-ubuntuvm ~/software/maverick]$ sudo apt remove python
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 ros-eloquent-rmw-implementation : Depends: ros-eloquent-rmw-fastrtps-cpp but it is not going to be installed or
                                            ros-eloquent-rmw-connext-cpp but it is not going to be installed or
                                            ros-eloquent-rmw-cyclonedds-cpp but it is not going to be installed or
                                            ros-eloquent-rmw-opensplice-cpp but it is not going to be installed
 ros-eloquent-rosidl-generator-c : Depends: ros-eloquent-rosidl-cmake but it is not going to be installed
 ros-eloquent-rosidl-parser : Depends: ros-eloquent-rosidl-adapter but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

Trying a fresh bootstrap with the new symlink.

fnoop, 2020-02-03:
symlink manifest order error:
```
Notice: /Stage[main]/Base::Python/File[/etc/profile.d/99-maverick-python-path.sh]/ensure: defined content as '{md5}d4781283425997b39908b4f81330bcd8'
Error: Could not set 'link' on ensure: No such file or directory @ dir_chdir - /srv/maverick/software/python/bin (file: /home/temp/maverick/manifests/maverick-modules/base/manifests/python.pp, line: 77)
Error: Could not set 'link' on ensure: No such file or directory @ dir_chdir - /srv/maverick/software/python/bin (file: /home/temp/maverick/manifests/maverick-modules/base/manifests/python.pp, line: 77)
Wrapped exception:
No such file or directory @ dir_chdir - /srv/maverick/software/python/bin
Error: /Stage[main]/Base::Python/File[/srv/maverick/software/python/bin/python]/ensure: change from 'absent' to 'link' failed: Could not set 'link' on ensure: No such file or directory @ dir_chdir - /srv/maverick/s
oftware/python/bin (file: /home/temp/maverick/manifests/maverick-modules/base/manifests/python.pp, line: 77)
Notice: /Stage[main]/Python::Install/Package[python]/ensure: created
```

fnoop, 2020-02-03:
So full bootstrap goes fine with the new python link.  All the base maverick scripts have been migrated to python3, all the pip installs have been migrated to python3.  What's left is whatever system packages are installed that are either python2 specific, or call python2 dependencies.
ROS is biggest problem.  Need to try and refactor with the python3 packages.

fnoop, 2020-02-05:
Looks like where ROS is installed by system packages, it requires python2 - in fact all of the ros packages require python2.  So removing python2 isn't an option.
Oddly ROS2 (ros-eloquent) breaks removing python2 as well - I thought ros2 was purely python3.

Anyway, closing for now.  The default `python` and `python3` commands both point to our custom python 3.7, and all maverick python pips are installed to the custom python3.7

closed 2020-02-05 by fnoop