goodrobots/maverick #830: visiond service on RPI Zero failing

HSVILJOEN, 2019-04-22:
Apr 22 13:24:13 maverick-raspberrylite maverick-visiond[2077]: Starting maverick-visiond
Apr 22 13:24:14 maverick-raspberrylite systemd[1]: maverick-visiond.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 13:24:14 maverick-raspberrylite systemd[1]: maverick-visiond.service: Unit entered failed state.
Apr 22 13:24:14 maverick-raspberrylite systemd[1]: maverick-visiond.service: Failed with result 'exit-code'.
Apr 22 13:24:44 maverick-raspberrylite systemd[1]: maverick-visiond.service: Service hold-off time over, scheduling restart.
Apr 22 13:24:44 maverick-raspberrylite systemd[1]: Stopped Maverick Vision Daemon.

------ Config -----
[visiond]

# On/off switch for maverick-visiond, set to False to disable
enable = True

# Camera device, almost always /dev/video0
# camera_device = /dev/video0

# Camera format, if not specified here will autodetect yuv->mjpeg->h264
# format = yuv

# Pixel Format, could be (fourcc)YUV2, I420, RGB etc
# pixelformat = I420

# Encoder, if not specified will default to h264.  Values are h264, mjpeg or none
# encoder = h264

# Resolution width of video stream, must be valid for camera
width = 1280

# Resolution height of video stream, must be valid for camera
height = 720

# Framerate of video stream, must be valid for camera
framerate = 30

# Brightness - 0 is automatic
brightness = 0

# Stream input type: v4l2 (fpv), appsrc (cv)
input = v4l2

# Stream output type: file (save video), udp (stream video), wcast (wifibroadcast), rtsp (rtsp server)
output = rtsp

# Output destination: filename (file output), IP address (udp output), Interface (wcast output)
output_dest = 0.0.0.0
# output_dest = wcast0

# Output port: Port number (eg. 5000) for network destination, Channel for wifibroadcast output (eg. 1)
output_port = 5600
# output_port = 1

# Pipeline Override - This is used to provide a manual pipeline if the auto construction fails o
# pipeline_override = v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! autovideoconvert ! vaapih264enc ! h264parse ! rtph264pay config-interva$
# pipeline override for Flir ONE colourized thermal stream
# v4l2src device=/dev/video3 ! queue ! autovideoconvert ! omxh264enc

# Debug: Turns on gstreamer debug to the specified level.  Note level 4 and above is very verbose
 debug = 5

# Hostname override for webvision, if detected wrongly by the system.  Can be IP address.
# webvision_hostname = maverick-raspberry.local

fnoop, 2019-05-01:
Hi @HSVILJOEN could you please attach a log file from ~/var/log/vision?

fnoop, 2019-09-21:
Confirmed working with maverick 2.0, will release a beta soon.

closed 2019-09-21 by fnoop