goodrobots/maverick #848: visiond service not streaming with RPi Zero

stalnikar, 2019-07-19:
Ref: Gitter chat with @fnoop 

We were initially unable to get the telnet connection to work with the RPi0 and had to run the `iptables -F` to resolve it. Once the connection was established, trying to stream visiond with VLC on a RPi 0 runs into kernel oops (#805) with the following log:

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.069030] Internal error: Oops: 805 [#1] ARM

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.119088] Process v4l2-source:src (pid: 1168, stack limit = 0xcd0ca188)

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.121336] Stack: (0xcd0cbc70 to 0xcd0cc000)

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.124216] bc60: 00000000 cd1a1200 00000001 ce96b4b0

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.129123] bc80: bf0e2e00 00000020 00000003 00000001 00000000 ce96b4b0 00000001 ce991d04

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.133365] bca0: 00000001 cd0cbd20 cd0cbcd0 cc0b6e60 cd0cbd14 cd0cbcc0 bf0de05c bf0dd8f4

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.137658] bcc0: cd0cbcd0 cd0cbcd0 00000001 00000003 00151800 cd0cbe18 cd0cbe18 cc251c40

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.141996] bce0: bf108a08 00000011 00000000 cc0b6e60 cd0cbd20 ce991d00 ce96b4b0 cd0cbd20

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.146700] bd00: 0000005c 00000000 cd0cbd5c cd0cbd18 bf0ea0f0 bf0ddf84 cd0cbd20 cc251c54

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.151984] bd20: 00151800 cd0cbe18 cd0cbe18 cc251c40 bf108a08 00000011 00000000 cc0b6e60

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.157560] bd40: ce991d00 ce96b058 cc0b6e60 bf108a08 cd0cbd7c cd0cbd60 bf0ea66c bf0ea00c

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.163267] bd60: bf0ea5e0 ce991d00 cc0b6e60 cc251c40 cd0cbd9c cd0cbd80 bf0a7f58 bf0ea5ec

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.168442] bd80: cc251c40 00000003 c0f8565c ce96b058 cd0cbe04 cd0cbda0 bf0a6ac8 bf0a7f04

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.173763] bda0: a96bd000 cff38eb8 ce8c5630 cc251c40 00000170 00000000 bf0bda60 ce991d00

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.179431] bdc0: bf0a6828 c0133f64 cd0cbe04 cd0cbdd8 c0133f64 ce991d00 ce991d00 000000f8

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.185554] bde0: ce991d00 ce991d00 c0f8565c cc0b6e60 bf0a6828 000000f8 cd0cbec4 cd0cbe08

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.192261] be00: bf0a64f8 bf0a6834 cd317370 00000000 00000003 ac7fe7e4 0000001b 00000001

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.198190] be20: 00151800 00002001 00000000 00000253 000de1a9 00000000 00000000 00000000

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.203953] be40: 00000000 00000000 00000001 023a6000 00151800 00000000 00000000 cd317370

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.209509] be60: b608b000 00000054 cd07ad80 cd30a22c cc20f894 00000000 cd0cbea4 cd0cbfb0

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.215643] be80: cd065f60 80000007 b608befc cc20f860 cc20f894 c05e3e1c cd0cbefc ce96b058

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.221418] bea0: cc0b6e60 ac7fe7e4 c0f8565c ce96b348 cd0ca000 00000000 cd0cbed4 cd0cbec8

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.228103] bec0: bf0a6820 bf0a62d0 cd0cbefc cd0cbed8 bf0a0688 bf0a6810 bf0a05b4 ac7fe7e4

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.233553] bee0: ceb405c8 cc0b6e60 0000001e 0000001e cd0cbf7c cd0cbf00 c0157a58 bf0a05c0

Message from syslogd@maverick-raspberrylite at Jul 15 18:56:28 ...
kernel:[ 596.239048] bf00: cd0cbf24 c0070ae0 cd0cbf5c cd0cbf18 c0070ae0 c04a9e7c 23bc3fbb 00000000

Message from syslogd@maverick-raspberrylite at Jul 15 18:

fnoop, 2019-08-10:
Related to #830?

Building new raspbian buster based build, possible/probable new kernel will fix this

fnoop, 2019-09-21:
Confirmed working with maverick 2.0.  Will release a beta version soon.

closed 2019-09-21 by fnoop