goodrobots/maverick #851: The configuration passes incorrectly. Some packages are missing.

hdw0, 2019-08-06:
Configuration passes with some errors. Dependency errors and missing packages. Included in the file. 
The ROS system does not work. I can run roscore, mavlink, etc. 
Maverick 1.0 is rarely a lot of mistakes and incorrect dependencies key packages. 

UBUNTU 16.04
Thanks ...
[error.txt](https://github.com/goodrobots/maverick/files/3474228/error.txt)

fnoop, 2019-08-06:
Try running `apt update` before doing a configure - that's the cause of most of the errors in this logfile.

fnoop, 2019-08-06:
> The ROS system does not work. I can run roscore, mavlink, etc.
Sweeping statement with no detail, gives nothing to look at.

fnoop, 2019-08-06:
> Maverick 1.0 is rarely a lot of mistakes and incorrect dependencies key packages.
..

fnoop, 2019-08-06:
What system is this running on?  Was this bootstrapped from scratch, or a downloaded image?

hdw0, 2019-08-07:
> Try running `apt update` before doing a configure - that's the cause of most of the errors in this logfile.

Yeah, I tried to do that. But at the same time there are some errors too. The file is attached.
[config_log.txt](https://github.com/goodrobots/maverick/files/3475360/config_log.txt)
[update_log.txt](https://github.com/goodrobots/maverick/files/3475361/update_log.txt)

hdw0, 2019-08-07:
I tried to install the system in different ways. the above log files are made on a virtual machine downloaded from this resource.

hdw0, 2019-08-07:
> What system is this running on? Was this bootstrapped from scratch, or a downloaded image?

Yes, bootstrapped from scratch.
Downloaded image work well, but cannot configure. (maverick 1.0) Later I can give the log file.

fnoop, 2019-08-07:
Your ROS errors are because the ROS key was dumped due to a security issue (with the central ROS repo).  Fix the key like this:
https://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/

I'll try and add this in automatically to the manifests.

hdw0, 2019-08-07:
Оn another VMware system in Ubuntu 16.04 Maverick 2.0 from bootstrap. Other problems, but also there is a connection with ROS. "bash: /opt/ros/kinetic/setup.bash: No such file or directory".

Also in this system, in graphical mode, the configuration is abruptly interrupted and the system is rebooted, restarting gives nothing. Only switching to text mode (multi-user.target) allows you to complete the configuration. Interrupted log in attach. 
Thanks ...

[config_log.txt](https://github.com/goodrobots/maverick/files/3476233/config_log.txt)

hdw0, 2019-08-07:
Same system, but in text mode. Configuration log.
[config_log_text_mode.txt](https://github.com/goodrobots/maverick/files/3476367/config_log_text_mode.txt)

fnoop, 2019-08-07:
Maverick 2.0 is really in alpha status, if you're expecting a smooth experience then it's not for you :)

It's long overdue, I've just started working on trying to get a beta release out soonish

hdw0, 2019-08-07:
> Maverick 2.0 is really in alpha status, if you're expecting a smooth experience then it's not for you :)
> 
> It's long overdue, I've just started working on trying to get a beta release out soonish

Before version 2.0 I had 3 days to install 1.0. I ran into a lot of problems ... Decided to try 2.0 and became something goes. Thanks for your help. 
I now try to return to the stable version. 
Maybe my job will help ...

fnoop, 2019-08-07:
v1 - the current stable (think 1.1.6 is the latest) branch should be pretty stable, lots of people have used this version.  Try starting with this and address each problem step by step.  If you can, log into gitter channel as we can help you in realtime there.
https://gitter.im/goodrobots/maverick

fnoop, 2019-08-09:
Various fixes applied to stable branch to fix repos/packages referenced by 1.1.6 that have since changed.

closed 2019-08-09 by fnoop