goodrobots/maverick #860: Lock down grafana version

fnoop, 2019-08-10:
If we're going to keep grafana, then at least lock down the package version.  At the moment we specify a version (because the api keeps changing), but then apt keeps updating it to the latest.  Either pin the pkg, build from source, or better yet just remove grafana.

SamuelDudley, 2019-08-10:
I quite like the graphics it produces visually but if I recall you were having issues taking the elements and displaying them in a custom view?

I'm personally willing to explore other options if you would like to drop it.

fnoop, 2019-09-14:
Use apt-mark to hold the version

fnoop, 2019-09-14:
Added bit of code but commented out, to hold grafana package, as it causes a puppet loop as grafana module can't cope with held packages.

Not worth the effort, grafana will probably disappear and this is an annoyance.

closed 2019-09-14 by fnoop