goodrobots/maverick #869: Implement basic monitoring system

fnoop, 2019-08-27:
A basic monitoring system will be very useful:
 - Monitor basic system parameters - CPU, disk space, memory etc
 - Monitor basic flight parameters - dangerous config, battery use/level, inconsistencies (eg. one motor working too hard)

SamuelDudley, 2019-08-28:
Currently what is used to obtain this info when populating the time series database? Can we interface that with the -api to allow the numbers to be viewable on a graphql driven dashboard?

fnoop, 2019-08-28:
collectd -> influxd.  But this is quite a heavy solution both storage and memory/cpu-wise, so shouldn't be mandatory or a dependency on alerts/monitoring, imho.  There are lots of opensource monitoring systems we can use.
For example, checkmk (which is kind of an uber-wrapper around other monitoring systems like nagios) has a query language like system for monitoring status:
https://checkmk.com/cms_livestatus.html
That would be kind of cool I think?

SamuelDudley, 2019-08-29:
Yeah, looks simple to interface with. Happy to roll with that as an option.