goodrobots/maverick #879: Python configuration management (again)

SamuelDudley, 2019-09-16:
It would be beneficial to allow elements of the system to create and use a virtual environment, however I'm not sure how possible this is with puppet....

e.g. for the -api a suitable virtual environment can be made:
`/srv/maverick/software/python/bin/python3 -m venv .venv`
`source .venv/bin/activate`
`pip install -r requirements.txt`
`pip install -r requirements-dev.txt`

This removes control from puppet and relies on the github tagged version of requirements.txt to be correct (which is okay IMO). Just looking at options to reduce the manifests updates and keep the versions inline with requirements.txt

fnoop, 2019-09-17:
Yes maverick does support python virtual environments, eg:
https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/maverick_fc/manifests/init.pp#L36

However they are mostly disabled/unused now - the reason is that Maverick tends to be a single-use environment and I didn't see the advantage of using them outweighed the complexity?

We now have a custom python environment in /srv/maverick/software/python that we use in preference for just about everything in the maverick environment.  If we get python module conflicts then it might be a good idea - if it's just about using requirements.txt then I've already started doing this:
https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_web/manifests/maverick_api.pp#L14

SamuelDudley, 2019-09-17:
It was more about using the requirements.txt and avoiding trying to keep it in sync with the manifests and having a single source of truth. Looks like you are doing that, cool!

closed 2019-09-17 by fnoop