goodrobots/maverick #883: Docker for SITL components?

SamuelDudley, 2019-10-08:
@fnoop  Apart from the downside of the overhead associated with running Docker itself, what are the disadvantages (only considering the SITL environment here which includes the autopilot and the associated physics engine)?

The main advantage of having SITL in a Docker container is the ability to launch the same configuration outside of Maverick. I only ask this question as I'm about to head down the road of building an image for the ArduPilot + MAVSDK testing.

fnoop, 2019-10-08:
Dockers are stateless - that is by design they're not supposed to hold or store data.  As I understand they're really meant to create lots of front-end webservers or apps, that connect to separate data stores.  So they're designed to be short-lived or essentially immutable - think 'throwaway'.

And you still have to write the docker files containing the same steps as you otherwise would have to write without a docker.  Then you have the overhead and extra complexity of docker itself, which is not inconsiderable.

For single-use embedded platforms like UAVs, my takeaway from looking at Docker was square peg / round hole - wrong use case.  For CI they may well be more suitable.

SamuelDudley, 2019-10-08:
Cool, appreciate the input. I'll likely borrow many of your puppet scripts setting up the Docker image and nothing else it will make for an interesting datapoint.

closed 2019-10-08 by SamuelDudley

fnoop, 2019-10-08:
Cool!