goodrobots/maverick #885: Initial configure fails demanding easy_install

hflemmen, 2019-11-30:
Hi!

We are trying to install Maverick on the Odroid XU4, but fails in the configure step. We were trying to follow the instructions from https://goodrobots.github.io/maverick/current/#/?id=bootstrap-maverick. 

The command we run:
```
maverick --env=dev configure
```

The error message we get:
```
Error: 'sudo easy_install -U pip==9.0.3' returned 1 instead of one of [0]
Error: /Stage[main]/Base::Python/Exec[upgrade-pip]/returns: change from 'notrun' to ['0'] failed: 'sudo easy_install -U pip==9.0.3' returned 1 instead of one of [0]
```

After this, the OS crashes and leaves an empty console with a blinking '_' in the upper left corner.

I have added the full log of two of our atempts.
[maverick.dev.2019-11-30-12-24.log](https://github.com/goodrobots/maverick/files/3907239/maverick.dev.2019-11-30-12-24.log)
[maverick.dev.2019-11-30-14-50.log](https://github.com/goodrobots/maverick/files/3907240/maverick.dev.2019-11-30-14-50.log)

Thank you for any tips in advance!

fnoop, 2019-11-30:
Hi @hflemmen - I would recommend that you try the master branch.  The stable branch is very old - I started a '2.0' release well over a year ago that was a bit too ambitious!

The master branch compiles and uses it's own python3.  You might hit a few problems with the odroid support as it was done with the 16.04 release and contained various hacks and fixes for the limitations of the odroid ubuntu at that time (which had many problems).  It might be that we need to disable the odroid hardware stuff altogether, but see how you get on with the master branch.

hflemmen, 2019-11-30:
Hi, thanks for the quick reply!

I tried with the master branch, but I did not get further. From the same command, I got this error:

```
Error: Evaluation Error: Error while evaluating a Resource Statement, Duplicate declaration: File[/srv/maverick/var/build/.install_flag_ros] is already declared at (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 285); cannot redeclare (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros2.pp, line: 248) (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros2.pp, line: 248, column: 13) on node odroid
``

fnoop, 2019-11-30:
Oops sorry that's a stupid mistake, if you do a `maverick self-update` and try again, should be fixed now :)

hflemmen, 2019-11-30:
Hi, thanks again for the quick reply!

I am still not able to run it through. It seems like it is struggling with ROS and ROS2.

I get errors like:
```
Error: Could not find command '/usr/bin/catkin'
Error: /Stage[main]/Maverick_ros::Ros1/Exec[ros-mavros-catkin-init]/returns: change from 'notrun' to ['0'] failed: Could not find command '/usr/bin/catkin'

...

Error: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install ros-melodic-ddynamic-reconfigure' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package ros-melodic-ddynamic-reconfigure
Error: /Stage[main]/Maverick_ros::Ros1/Package[ros-melodic-ddynamic-reconfigure]/ensure: change from 'purged' to 'present' failed: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install ros-melodic-ddynamic-reconfigure' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package ros-melodic-ddynamic-reconfigure
```

I have attached two of the logs from my atempts: 
[maverick.dev.2019-11-30-18-45.log](https://github.com/goodrobots/maverick/files/3907465/maverick.dev.2019-11-30-18-45.log)
[maverick.dev.2019-11-30-19-14.log](https://github.com/goodrobots/maverick/files/3907466/maverick.dev.2019-11-30-19-14.log)

Thanks again!

fnoop, 2019-11-30:
Oops yes, I haven't tested source installs of the new ros1/ros2 multi environment yet..  I'll test and upload fixes.

hflemmen, 2019-11-30:
Thank you! Say if there is anything we can do. :)

hflemmen, 2019-12-01:
For the planning of our project, do you have a time estimate for when you have time to test it?

fnoop, 2019-12-03:
Hi @hflemmen I'm busy until at least the end of this week, but I'll have a lot more time after that.  I'll try and fully test the source build by the weekend.

hflemmen, 2019-12-03:
Thanks for the update! We are eagerly awaiting the fix.

hflemmen, 2019-12-14:
Hi again!
How did the source build turn out?

fnoop, 2019-12-14:
Hey, sitting here just reinstalling jetson nano from scratch to find out :)

fnoop, 2019-12-16:
Turns out lots of bootstrap problems with latest master.  Will track and fix using this issue.

First up:
```
Notice: /Stage[main]/Maverick_ros::Ros1/File[/srv/maverick/software/maverick/bin/mavros.sh]/ensure: created
Error: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Error: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Wrapped exception:
No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock
Error: /Stage[main]/Maverick_ros::Ros1/File[/opt/ros/melodic/share/mavros/launch/apm_config.yaml]/ensure: change from 'absent' to 'file' failed: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Notice: /Stage[main]/Maverick_ros::Ros2/File[/srv/maverick/software/ros2]/ensure: created
```

The package  `ros-melodic-ddynamic-reconfigure` exists and installs fine, so assume this fragment of the manifest is trying to install before the repo is installed/updated.

fnoop, 2019-12-16:
```
Notice: /Stage[main]/Maverick_ros::Ros1/File[/srv/maverick/software/maverick/bin/mavros.sh]/ensure: created
Error: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Error: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Wrapped exception:
No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock
Error: /Stage[main]/Maverick_ros::Ros1/File[/opt/ros/melodic/share/mavros/launch/apm_config.yaml]/ensure: change from 'absent' to 'file' failed: Could not set 'file' on ensure: No such file or directory @ dir_s_mkdir - /opt/ros/melodic/share/mavros/launch/apm_config.yaml20191216-2389-vbitx7.lock (file: /srv/maverick/software/maverick/manifests/maverick-modules/maverick_ros/manifests/ros1.pp, line: 499)
Notice: /Stage[main]/Maverick_ros::Ros2/File[/srv/maverick/software/ros2]/ensure: created
```

fnoop, 2019-12-16:
ros2  pk error:

```
Notice: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]/returns: Err:10 http://packages.ros.org/ros2/ubuntu bionic InRelease
Notice: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]/returns:   The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Notice: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]/returns: Reading package lists...
Notice: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]/returns: W: GPG error: http://packages.ros.org/ros2/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Notice: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]/returns: E: The repository 'http://packages.ros.org/ros2/ubuntu bionic InRelease' is not signed.
Error: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]: Failed to call refresh: '/usr/bin/apt update' returned 100 instead of one of [0]
Error: /Stage[main]/Maverick_ros::Ros2/Exec[ros2_apt_update]: '/usr/bin/apt update' returned 100 instead of one of [0]
Notice: /Stage[main]/Maverick_ros::Ros2/Package[ros-dashing-ros-base]: Dependency Exec[ros2_apt_update] has failures: true
```

fnoop, 2019-12-16:
mavsdk build error:
```
Notice: /Stage[main]/Maverick_dev::Mavsdk/Exec[mavsdk-prepbuild]/returns: executed successfully
Error: '/usr/bin/make -j2 >/srv/maverick/var/log/build/mavsdk.build.out 2>&1' returned 2 instead of one of [0]
Error: /Stage[main]/Maverick_dev::Mavsdk/Exec[mavsdk-build]/returns: change from 'notrun' to ['0'] failed: '/usr/bin/make -j2 >/srv/maverick/var/log/build/mavsdk.build.out 2>&1' returned 2 instead of one of [0]
Notice: /Stage[main]/Maverick_dev::Mavsdk/Exec[mavsdk-install]: Dependency Exec[mavsdk-build] has failures: true
Warning: /Stage[main]/Maverick_dev::Mavsdk/Exec[mavsdk-install]: Skipping because of failed dependencies
Warning: /Stage[main]/Maverick_dev::Mavsdk/File[/srv/maverick/var/build/.install_flag_mavsdk]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_dev::Ardupilot/Oncevcsrepo[git-ardupilot]/File[/srv/maverick/code/ardupilot]/ensure: created
```

fnoop, 2019-12-18:
Mostly clean build now on x86_64 and arm64 (nvidia jetson nano).  Few remaining issues:

```
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: -- Could NOT find nodelet (missing: nodelet_DIR)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: -- Could not find the required component 'nodelet'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   Could not find a package configuration file provided by "nodelet" with any
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   of the following names:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:     nodeletConfig.cmake
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:     nodelet-config.cmake
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   Add the installation prefix of "nodelet" to CMAKE_PREFIX_PATH or set
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   "nodelet_DIR" to a directory containing one of the above files.  If
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   "nodelet" provides a separate development package or SDK, be sure it has
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   been installed.
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Call Stack (most recent call first):
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:   realsense-ros/realsense2_camera/CMakeLists.txt:8 (find_package)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: -- Configuring incomplete, errors occurred!
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: See also "/srv/maverick/var/build/catkin_ws_realsense/build/CMakeFiles/CMakeOutput.log".
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: See also "/srv/maverick/var/build/catkin_ws_realsense/build/CMakeFiles/CMakeError.log".
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Base path: /srv/maverick/var/build/catkin_ws_realsense
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Source space: /srv/maverick/var/build/catkin_ws_realsense/src
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Build space: /srv/maverick/var/build/catkin_ws_realsense/build
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Devel space: /srv/maverick/var/build/catkin_ws_realsense/devel
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Install space: /srv/maverick/var/build/catkin_ws_realsense/install
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: ####
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: #### Running command: "cmake /srv/maverick/var/build/catkin_ws_realsense/src -DCATKIN_DEVEL_PREFIX=/srv/maverick/var/build/catkin_ws_realsense/devel -DCMAKE_INSTALL_PREFIX=/srv/maverick/var/build/catkin_ws_realsense/install -G Unix Makefiles" in "/srv/maverick/var/build/catkin_ws_realsense/build"
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: ####
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: Invoking "cmake" failed
Error: '/opt/ros/current/bin/catkin_make clean; /opt/ros/current/bin/catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/srv/maverick/software/ros/melodic >/srv/maverick/var/log/build/ros.realsense.catkin_make.out 2>&1' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-make]/returns: change from 'notrun' to ['0'] failed: '/opt/ros/current/bin/catkin_make clean; /opt/ros/current/bin/catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/srv/maverick/software/ros/melodic >/srv/maverick/var/log/build/ros.realsense.catkin_make.out 2>&1' returned 1 instead of one of [0]
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-install]: Dependency Exec[ros-realsense-catkin-make] has failures: true
Warning: /Stage[main]/Maverick_ros::Ros1/Exec[ros-realsense-catkin-install]: Skipping because of failed dependencies
```

fnoop, 2019-12-18:
```
Notice: /Stage[main]/Maverick_analysis::Mavlogd/Service[maverick-mavlogd]/ensure: ensure changed 'stopped' to 'running'
Error: Systemd start for maverick-nginx failed!
journalctl log for maverick-nginx:
-- Logs begin at Mon 2019-12-16 10:51:57 GMT, end at Wed 2019-12-18 15:26:02 GMT. --
Dec 18 15:25:59 maverick-ubuntu systemd[1]: Starting The NGINX HTTP and reverse proxy server...
Dec 18 15:25:59 maverick-ubuntu nginx[8771]: nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
Dec 18 15:25:59 maverick-ubuntu nginx[8771]: nginx: configuration file /etc/nginx/nginx.conf test is successful
Dec 18 15:25:59 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:00 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:00 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:01 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:01 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:02 maverick-ubuntu nginx[8772]: nginx: [emerg] still could not bind()
Dec 18 15:26:02 maverick-ubuntu systemd[1]: maverick-nginx.service: Control process exited, code=exited status=1
Dec 18 15:26:02 maverick-ubuntu systemd[1]: maverick-nginx.service: Failed with result 'exit-code'.
Dec 18 15:26:02 maverick-ubuntu systemd[1]: Failed to start The NGINX HTTP and reverse proxy server.

Error: /Stage[main]/Nginx::Service/Service[maverick-nginx]/ensure: change from 'stopped' to 'running' failed: Systemd start for maverick-nginx failed!
journalctl log for maverick-nginx:
-- Logs begin at Mon 2019-12-16 10:51:57 GMT, end at Wed 2019-12-18 15:26:02 GMT. --
Dec 18 15:25:59 maverick-ubuntu systemd[1]: Starting The NGINX HTTP and reverse proxy server...
Dec 18 15:25:59 maverick-ubuntu nginx[8771]: nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
Dec 18 15:25:59 maverick-ubuntu nginx[8771]: nginx: configuration file /etc/nginx/nginx.conf test is successful
Dec 18 15:25:59 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:00 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:00 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:01 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:01 maverick-ubuntu nginx[8772]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Dec 18 15:26:02 maverick-ubuntu nginx[8772]: nginx: [emerg] still could not bind()
Dec 18 15:26:02 maverick-ubuntu systemd[1]: maverick-nginx.service: Control process exited, code=exited status=1
Dec 18 15:26:02 maverick-ubuntu systemd[1]: maverick-nginx.service: Failed with result 'exit-code'.
Dec 18 15:26:02 maverick-ubuntu systemd[1]: Failed to start The NGINX HTTP and reverse proxy server.

Notice: /Stage[main]/Nginx::Service/Service[maverick-nginx]: Triggered 'refresh' from 3 events
Notice: /Stage[main]/Maverick_web::Maverick_docs/Nginx::Resource::Location[web-maverick-docs]/Concat::Fragment[maverick-ubuntu.tendawifi.com-500-e0a27f41fba4fbc2bdf311b78866dd26]/Concat_fragment[maverick-ubuntu.tendawifi.com-500-e0a27f41fba4fbc2bdf311b78866dd26]: Dependency Service[maverick-nginx] has failures: true
```

hflemmen, 2019-12-18:
Nice! I can maybe test it on Thursday. If not, I can test it next year.

fnoop, 2019-12-19:
Arm tensorflow missing:
```
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: Collecting tensorflow==1.14.0
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns:   ERROR: HTTP error 404 while getting https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns:   ERROR: Could not install requirement tensorflow==1.14.0 from https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow because of error 404 Client Error: Not Found for url: https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl
Notice: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: ERROR: Could not install requirement tensorflow==1.14.0 from https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow because of HTTP error 404 Client Error: Not Found for url: https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl for URL https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow
Error: 'pip3 --log /tmp/pip.log install   --disable-pip-version-check  'https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow'' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_intelligence::Tensorflow/Install_python_module[tensorflow-pip]/Python::Pip[tensorflow-pip]/Exec[pip_install_tensorflow-pip]/returns: change from 'notrun' to ['0'] failed: 'pip3 --log /tmp/pip.log install   --disable-pip-version-check  'https://github.com/PINTO0309/Tensorflow-bin/raw/master/tensorflow-1.14.0-cp37-cp37m-linux_aarch64.whl#egg=tensorflow'' returned 1 instead of one of [0]
```

fnoop, 2019-12-19:
Codeserver fails on jetson nano, because it doesn't support arm yet:
```
Notice: /Stage[main]/Maverick_web::Maverick_web/Package[yarn]/ensure: created
Error: 'yarn build 1.41.0 production >/srv/maverick/var/log/build/codeserver.build.log 2>&1' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_web::Codeserver/Exec[codeserver-build]/returns: change from 'notrun' to ['0'] failed: 'yarn build 1.41.0 production >/srv/maverick/var/log/build/codeserver.build.log 2>&1' returned 1 instead of one of [0]
Notice: /Stage[main]/Maverick_web::Codeserver/Exec[codeserver-package]: Dependency Exec[codeserver-build] has failures: true
```

Need to restrict codeserver builds to x86_64.

hflemmen, 2019-12-19:
I ran `maverick self-update` and still it crashed under `maverick --env=dev configure`.

I got a new error:
`Error: Could not find command '/usr/bin/catkin'
Error: /Stage[main]/Maverick_ros::Ros1/Exec[ros-mavros-catkin-init]/returns: change from 'notrun' to ['0'] failed: Could not find command '/usr/bin/catkin'`

hflemmen, 2019-12-19:
Very nice with the quick response! I would like to, but unfortunately, I just left the lab. I probably won't be able to test it on the Odroid XU4 until next year. I could maybe try it on a Pi3, but not on the Odroid.

hflemmen, 2020-01-07:
Hi again, happy new year!

I ran `maverick self-update`, and tried `maverick --env=dev configure` again, but also this time it crashed. It seems to be the same error:
```
Error: Could not find command '/usr/bin/catkin'
Error: /Stage[main]/Maverick_ros::Ros1/Exec[ros-opencv-catkin-init]/returns: change from 'notrun' to ['0'] failed: Could not find command '/usr/bin/catkin'
```
Log:
[maverick.dev.2020-01-07-15-18.log](https://github.com/goodrobots/maverick/files/4031332/maverick.dev.2020-01-07-15-18.log)

This is still on the Odroid XU4 with Ubuntu Mate 18.04.

fnoop, 2020-01-07:
Happy new year :)

Ah yes this is a source ros build.  I've added another dep - you can test again, i'll force a source build at this end and try again.

hflemmen, 2020-01-07:
Thanks for the quick reply!

I updated and got one addition. When I tried to configure, I got one new error:
```
Error: Could not find command '/srv/maverick/software/python/bin/vcs'
Error: /Stage[main]/Maverick_ros::Ros2/Exec[ros2-vcs-import]/returns: change from 'notrun' to ['0'] failed: Could not find command '/srv/maverick/software/python/bin/vcs'
```
The full log:
[maverick.dev.2020-01-07-16-12.log](https://github.com/goodrobots/maverick/files/4031470/maverick.dev.2020-01-07-16-12.log)

fnoop, 2020-01-08:
I've tested ros2 source build and made a couple of commits to fix, try again now :)

hflemmen, 2020-01-09:
Hi again!

I  tried again, but some problem persists. The log contains no error, but the screen goes black and I have to powercycle to get it running again. This has happened all the times I have tried earlier, but then I thought it was because of the errors. Now I don't know.

Symptom: Screen turns black with a little '_' blinking in the upper left corner. It takes no input, and does not change state before I turn off the power.
When: A little while into the initial configure after the call to `maverick --env=dev configure`.

Two similar logs, but there is no fatal error in them:
[maverick.dev.2020-01-09-18-15.log](https://github.com/goodrobots/maverick/files/4041907/maverick.dev.2020-01-09-18-15.log)
[maverick.dev.2020-01-09-18-21.log](https://github.com/goodrobots/maverick/files/4041908/maverick.dev.2020-01-09-18-21.log)

I am somewhat confused. If you know what to try, it would be very interesting.

fnoop, 2020-01-10:
Hmm, that's odd.  Would recommend you head over to the gitter for more interactive help:
https://gitter.im/goodrobots/maverick

hflemmen, 2020-01-12:
The log when running `maverick --env=dev --debug configure`(with `--debug`):
[maverick.dev.2020-01-12-18-09.log](https://github.com/goodrobots/maverick/files/4050931/maverick.dev.2020-01-12-18-09.log)

hflemmen, 2020-01-17:
Hi again,

I finally had the time to run the whole thing.

For future reference:
The problem I had where the screen was black was because I was configuring from a desktop environment, while the maverick settings was to remove any desktop environments. My solution was to enable desktop environment in my localconf.json file.

Now that the process have been running a little longer, I got some more errors. The log is attaced below:
[maverick.dev.2020-01-14-15-14.log](https://github.com/goodrobots/maverick/files/4078829/maverick.dev.2020-01-14-15-14.log)

This was after disabling tensor flow which also crashed.

commit fd1fcdff (2019-12-19): More reordering fixes #885
closed 2020-07-14 by fnoop