goodrobots/maverick #894: visiond RTSP Not Working RPi Lite

cglusky, 2020-01-15:
Title probably needs to be changed but here is the setup:

Setup:
- RPi0W with Cam Module v1.3 (5mp/1080p)
- Fresh install of lite image
- self-update
- configure
- reboot
- stop/disable all services except FC, nginx; start/enable visiond (essential FPV setup)
- reboot
- Check default visiond.conf and all is good

Result:
- Not receiving rtsp  in QGC (Win10, MacOS) or MissionPlanner (Win10).  Win10 has firewall OFF.
- Not assuming zeroconf works so using IP address to target rtsp server.
- No errors in visiond log file.

Temp Fix:
- Switch to UDP and video stream works great.

Did I shut down a service that is a dependency for rtsp?  I left nginx running so routing would work etc.  And no errors in visiond log when rtsp is used.

cglusky, 2020-01-15:
Take that back.  I get the first few frames via UDP and then it locks up.  Don't have a heatsink on this 0 yet so will see if it's thermal issue.

cglusky, 2020-01-15:
It just seems to lock up after it sends a few frames.  There has to be a log but I can't find it.

fnoop, 2020-01-16:
https://github.com/goodrobots/maverick/issues/830
https://github.com/goodrobots/maverick/issues/848

Fixed in master a while back.  Will release a new  image shortly.

closed 2020-01-16 by fnoop

cglusky, 2020-01-16:
hah.  sorry.  let's see how many times i can repeat stuff for awhile back.