goodrobots/maverick #898: Test flight mode

fnoop, 2020-01-23:
Flight mode should turn off everything not needed for flight.

fnoop, 2020-01-23:
```
[flight] [mav@maverick-ubuntuvm ~]$ maverick status

Maverick - UAV Companion Computer System - Version 1.2.0-beta

Maverick Environment:    flight

Service              Description                Status | At Boot
------------         ------------              ------------------
____________
Web Services
webdev               Web Devserver             Stopped | Disabled
nginx                Webserver                 Running | Enabled
cloud9               Cloud9 IDE                Stopped | Disabled
codeserver           CodeServer/VSCode IDE     Running | Enabled
theia                Theia IDE                 Stopped | Disabled
____________
Analysis Services
grafana              Analysis Dashboard        Running | Enabled
influxd              TimeSeries Database       Running | Enabled
collectd             System Metrics Collector  Running | Enabled
mavlogd              Maverick Log Importer     Running | Enabled
uploader             Maverick Log Uploader     Running | Enabled
____________
Network Services
____________
Vision Services
visiond              Vision Service            Stopped | Disabled
camera-manager       Intel Camera Manager      Stopped | Disabled
mavcesium            MavCesium                 Stopped | Disabled
vision_landing       Precision Landing         Stopped | Disabled
vision_seek          Seek Thermal Vision       Stopped | Disabled
webvision            Web Vision                Running | Enabled
____________
Hardware Services
flirone              Flir One Vision           Stopped | Disabled
____________
Flight Controller
mavlink@fc           Mavlink (FC)              Stopped | Disabled
rosmaster@fc         Rosmaster (FC)            Stopped | Disabled
rosmaster@fc         Rosmaster (FC)            Stopped | Disabled
api@fc               MavAPI (FC)               Stopped | Disabled
____________
APSITL (dev)
apsitl@dev           AP SITL (dev)             Running | Enabled
mavlink@dev          Mavlink (dev)             Running | Enabled
rosmaster@dev        ROS (dev)                 Running | Enabled
mavros@dev           MavROS (dev)              Running | Enabled
api@dev              MavAPI (dev)              Running | Enabled
```

SITL still running

SamuelDudley, 2020-01-23:
Note FC is stopped and disabled.
I'm assuming there is a way of users to set the config for what is on / off during flight / dev?
If so I'll add that to the -web backlog at a lower priority.

fnoop, 2020-02-04:
Main problem was that most of the manifests and sample configs still related to maverick_dev::sitl class which was changed a while back to maverick_dev::apsitl_dev.  I've fixed all the manifests, config and docs.

So now without anything specific set in localconf.json, a `maverick configure --env=flight` will disable the SITL stuff and enable the flight stuff:
```
____________
Flight Controller
mavlink@fc           Mavlink (FC)              Stopped | Enabled
rosmaster@fc         Rosmaster (FC)            Running | Enabled
rosmaster@fc         Rosmaster (FC)            Running | Enabled
api@fc               MavAPI (FC)               Running | Enabled
____________
APSITL (dev)
apsitl@dev           AP SITL (dev)             Stopped | Disabled
mavlink@dev          Mavlink (dev)             Stopped | Disabled
rosmaster@dev        ROS (dev)                 Stopped | Disabled
mavros@dev           MavROS (dev)              Stopped | Disabled
api@dev              MavAPI (dev)              Stopped | Disabled
```
(mavlink@fc is stopped because the serial device doesn't exist in a VM)

Going back with `maverick configure --env=dev` disables the FC services and re-enables the SITL services:
```
____________
Flight Controller
mavlink@fc           Mavlink (FC)              Stopped | Disabled
rosmaster@fc         Rosmaster (FC)            Stopped | Disabled
rosmaster@fc         Rosmaster (FC)            Stopped | Disabled
api@fc               MavAPI (FC)               Stopped | Disabled
____________
APSITL (dev)
apsitl@dev           AP SITL (dev)             Running | Enabled
mavlink@dev          Mavlink (dev)             Running | Enabled
rosmaster@dev        ROS (dev)                 Running | Enabled
mavros@dev           MavROS (dev)              Running | Enabled
api@dev              MavAPI (dev)              Running | Enabled
```

Going to flight mode disables all IDEs, but you can't re-enable them because they're not 'set' elsewhere, because..

The problem with all of this is the hierarchical configuration.  In Maverick, the hierarchy is set in ~/conf/hiera.ymal.  It's set so that the basic lowest-level config is puppet-defaults.json and maverick-defaults.json which essentially just setup blank containers to ensure they exist, then the sample-nodes configs which are used to create the dist images, then the environment config.  It's these environment configs that we're using here to set --env and --flight mode config.  However the highest level config is the localconf.json, and so anything set in here will override everything else, including the environment config.  So if you have any of the FC or SITL services set in localconf.json it will override the environment configs, likewise with the IDE configs.  This can be confusing (very confusing, probably to the normal user), but it's necessary to allow the extensive flexibility that is inherent in the system - it's necessary for example to allow fleets of UAVs each with different roles, and overrides for individual UAVs.

fnoop, 2020-02-04:
So in short, this is fixed, but come the UI we will need to present the config and services very carefully to be as less confusing as possible.

closed 2020-02-04 by fnoop