goodrobots/maverick #900: Document all puppet manifest parameters

fnoop, 2020-01-23:
https://puppet.com/blog/using-puppet-strings-generate-great-documentation-puppet-modules/

SamuelDudley, 2020-01-23:
the blog post is a little old but it looks like the later versions support HTML, JSON and markdown as output targets

fnoop, 2020-01-23:
Need to install puppet-strings gem in bin/maverick

SamuelDudley, 2020-01-25:
https://github.com/puppetlabs/puppet-strings/blob/master/JSON.md

Example json output. Likely we only care about the parameters. Valid values can be supplied along with types.

fnoop, 2020-01-25:
Also use this as a kick to type the puppet parameters, which I've been too lazy to do up to now

fnoop, 2020-01-26:
https://puppet.com/blog/meet-puppet-strings-2/

fnoop, 2020-01-28:
To generate the puppet strings output:

`puppet strings generate **/*.pp —format markdown —out >/var/tmp/mavdoc.md`
or
`puppet strings generate **/*.pp —format json >/var/tmp/mavdoc.json`
or
`puppet strings generate **/*.pp`

Run those from ~/software/maverick/manifests/maverick-modules.  The latter will generate a 'docs' directory in ~/software/maverick/manifests/maverick-modules which contains the html generate docs - open index.html in a browser.

SamuelDudley, 2020-02-01:
possible issue with `maverick_ros::ros1::distribution`:
```
        "maverick_ros::ros1::distribution": {
            "description": "Specify the ROS distribution to use - eg. kinetic, melodic\n\n",
            "type": "string",
            "default": "auto"
        },
```
default is `auto`... is that correct?

SamuelDudley, 2020-02-01:
likewise for `maverick_ros::ros2::distribution`

fnoop, 2020-02-01:
Yes, if 'auto' it works out the ros distribution based on OS and architecture:
https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_ros/manifests/ros1.pp#L47

SamuelDudley, 2020-02-01:
`maverick_security::ldap_server::$server_type`
^ note` $`
```
        "maverick_security::ldap_server::server_type": {
            "description": "\n\n",
            "type": "string",
            "default": "openldap"
        },
```
also exists

SamuelDudley, 2020-02-01:
last one is more of a question than an issue... 
there are two types that I am not yet handling: 

- `Hash` 
- `Hash[String, Variant[String, Integer, Boolean, Hash[String, Variant[String, Integer, Boolean]]]]`

`Hash` belongs to `maverick_dev::ardupilot::ardupilot_all_vehicles` and appears to be a mapping with a default value of `{"copter" => "arducopter", "plane" => "arduplane", "rover" => "ardurover", "sub" => "ardusub", "heli" => "arducopter-heli", "antennatracker" => "antennatracker"}`. How is this used in the config.json file? Can you provide an example of what a user could enter for this parameter?

`Hash[String, Variant[String, Integer, Boolean, Hash[String, Variant[String, Integer, Boolean]]]]` belongs to `base::hostip::additional_entries`. The default value is `{}`. Is this type correct? Can you provide an example of what a user could enter for this parameter?

SamuelDudley, 2020-02-01:
looking at https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_dev/manifests/ardupilot.pp it appears that the hash is internal to the manifest: https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_dev/manifests/ardupilot.pp#L73

need to workout if a hash can be represented in a json config file... perhaps its an object with key, value pairs?

SamuelDudley, 2020-02-01:
I can confirm the type provided by puppet strings for `base::hostip::additional_entries` is correct.
https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/base/manifests/hostip.pp#L52

Once again, unsure how a user would add a value to `base::hostip::additional_entries`.
Closing this issue again as only parsing issues of the hash types remain (not documentation)

closed 2020-02-01 by SamuelDudley

fnoop, 2020-02-01:
> also exists, possible error?
BTW great job on the docs, just parsing everything and doing a sanity check. Hope you dont mind.

Oh yes I'm sure you'll find some errors, by the time I got halfway through the manifests my mind had turned to mush :)

fnoop, 2020-02-01:
> Once again, unsure how a user would add a value to base::hostip::additional_entries.
Closing this issue again as only parsing issues of the hash types remain (not documentation)

I'll be honest as I came to this I couldn't remember either!  It should be the same format as $host_entries and $_ip6_stdaliases, so something like the json representation of:
            "${_fqdn}" => {
                ip              => $_ipaddress,
                host_aliases    => $_host_aliases,
            }

Off the top of my head, network interfaces are declared in json as a hash of hashes, or at least an array of hashes, so you should be able to see how they're defined in there.

fnoop, 2020-02-01:
> need to workout if a hash can be represented in a json config file... perhaps its an object with key, value pairs?

Have a look at network interface definitions in the sample configs, eg.
~/software/maverick/conf/sample-nodes/maverick-raspberry.json