goodrobots/maverick #904: SSH Issues

ashwinv96, 2020-01-31:
Hi, 
I downloaded the image for RBPi 3 and tried to SSH in. Was not working with the error:
"ssh: Could not resolve hostname maverick-raspberry.local: Name or service not known"
I changed the hostname to mav@maverick-raspberry.local and still got same issue. 

For some reason every 1 in 50 attempts it successfully is able to SSH in and PING returns data. I really am very confused as to why this is happening and am frustrated :(

Anyone face something similar?

fnoop, 2020-01-31:
Hi, what OS are you trying to ssh in from?  Note that windows doesn't support zeroconf natively, so using the hostname won't work.
If you're using Linux/Mac, then it might be that Avahi running on the raspberry is using the wrong hostname for some reason.  If you can ssh in using the IP address (usually you can find it on your router), then look at the avahi process:
```
ps -ef |grep avahi
avahi      777     1  0 Jan29 ?        00:00:06 avahi-daemon: running [maverick-ubuntuvm.local]
```
The hostname at the end in brackets for some reason sometimes gets a suffix added, like 'maverick-raspberry-12.local'.  If you restart it, then it usually resets to the correct hostname, and then you can login again:
`sudo systemctl restart avahi-daemon`

fnoop, 2020-02-23:
Please reopen if you're still having problems.

closed 2020-02-23 by fnoop