goodrobots/maverick #925: hostname changed to wrong interface

fnoop, 2020-03-14:
`Notice: /Stage[bootstrap]/Hosts/Host[maverick-nano.tendawifi.com]/ip: ip changed '192.168.5.129' to '172.17.0.1'`

```
7: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 02:42:74:7a:55:12 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
```

Interface is down, shouldn't take IP from here.  Also shouldn't take from docker..

fnoop, 2020-03-23:
If  'primaryip' is set in hiera(localconf) then this is used, otherwise it falls back to the ipaddress fact.  Facter says this is deprecated:
https://puppet.com/docs/facter/3.9/core_facts.html#ipaddress
What should we use?

fnoop, 2020-03-23:
https://www.gesellix.net/post/facter-docker-and-the-public-ip-address/
https://tickets.puppetlabs.com/browse/FACT-380

fnoop, 2020-03-23:
Separate facter project is stuck at 3.11, but more recent versions are documented as part of puppet-opensource:
https://puppet.com/docs/puppet/latest/release_notes_facter.html#concept-5716

gem shows a later 4.0 version, but this isn't compatible with current puppet6 releases.  Use the latest version distributed with puppet.

fnoop, 2020-03-23:
With the new facter we get a warning:
`Warning: Could not retrieve either serverip or serverip6 fact`
Once this is resolved, maybe serverip is a better fact to use.

fnoop, 2020-03-23:
There is no facter 3.x available through gem, only 4.0.12 which doesn't exist anywhere in puppetlabs docs or in github releases.  Bit of a mess - backtrack to 2.5.7 for now ..

fnoop, 2020-03-23:
As per the FACT-380 notes, running docker will effectively break the ipaddress fact:
`Notice: /Stage[bootstrap]/Hosts/Host[maverick-nano.local]/ip: ip changed '192.168.1.91' to '172.17.0.1'`

Either we disable docker, or we find another workaround.

SamuelDudley, 2020-03-23:
Sorry for the impending dumb question: what is using Docker on a maverick image? Some of the configuration tooling?

fnoop, 2020-03-23:
No absolutely nothing - but it's installed on some platforms by default, or as a dependency for some things.  It popped up on my nano ubuntu, can't remember if it was there at install or not.

I'm going to disable/uninstall it by default, but with a parameter to turn it back on, and will document it along with the `primaryip` localconf parameter that can override.

fnoop, 2020-03-23:
Added `base::packages::docker` param, false by default, which if false disables and uninstalls docker including the docker0 interface.  However it needs `maverick configure` to be run twice - once to disable the interface, the second to pick up the new primary ip and write it to /etc/hosts.

fnoop, 2020-03-23:
Will have to do for now - keep open to document `primaryip` param and docker params.