goodrobots/maverick #939: webrtc stream fails first start

fnoop, 2020-03-30:
See https://github.com/goodrobots/maverick-web/issues/172

fnoop, 2020-03-30:
Changed boot order so webrtc starts after visiond in systemd startup, but needs more than that.  visiond takes time to startup the video stream and create a client, webrtc is starting before the stream starts, fails and doesn't retry.  Restarting webrtc later on correctly connects to the visiond rtsp stream and then works.

Need to somehow delay the webrtc startup until visiond rtsp stream is fully active.

SamuelDudley, 2020-03-30:
https://github.com/facebookincubator/pystemd/blob/master/_docs/daemon.md

We can use pystemd to notify systemd when the stream is created. Webrtc will wait until the notification occurs.

fnoop, 2020-03-30:
Think this is fixed now

SamuelDudley, 2020-03-31:
Interesting, this since your patch I have not been able to get a stream running. I can see that webrtc is trying to start but visiond was totally failing to start. Perhaps it's related to the http get in visond (janus thread) being made with webrtc not yet running? Need to have a closer look.

fnoop, 2020-03-31:
What is the visiond log saying?

SamuelDudley, 2020-03-31:
Visond is getting a connection refused error on line 86 of janus.py (the get_info request) which causes it to exit and retry. At the same time webrtc is in a loop not able to connect to the websocket handler (as visond has not started).

SamuelDudley, 2020-03-31:
Possibly we could delay the get_info request until after the websock server has started?

fnoop, 2020-03-31:
> Visond is getting a connection refused error on line 86 of janus.py (the get_info request) which causes it to exit and retry. At the same time webrtc is in a loop not able to connect to the websocket handler (as visond has not started).

Ah.  Odd I'm not getting that, but that's only there to query janus if it's already up and running, so just wrap it in a try/except.
visiond then starts up the websocket which janus connects to and fires events over.  The initial query is only for if janus is already running.

SamuelDudley, 2020-03-31:
Sure thing. Leave it with me.

fnoop, 2020-03-31:
Updated visiond, should work now

fnoop, 2020-03-31:
It's looking for the other sample files that maverick is removing - ignore those warnings.

SamuelDudley, 2020-03-31:
okay, turns out it was a mix of things..
`input = v4l2` is no longer a valid option in visiond and that was causing a failure. That was only a recent problem. The bigger problem was that the video streaming didnt work on linux + firefox or ios + chrome (ipad) so I was second guessing if webrtc was working as expected. Works fine (after fixing the visiond config file on linux + chrome.

closed 2020-03-31 by SamuelDudley

fnoop, 2020-04-01:
> input = v4l2 is no longer a valid option in visiond and that was causing a failure
oops, sorry - I normalised the inputs and changed all the configs, but forgot it would affect existing configs files.
The default now is to have a completely blank config and let everything autodetect.

SamuelDudley, 2020-04-01:
I just wiped the config file and all was well 👍 I think the `input = v4l2` option is still floating around in there, just commented out. Are there any other options that if uncommented would cause problems?

fnoop, 2020-04-01:
Oh yes, updated thanks!  I've been through all the other configs and changed all the defaults in visiond so I think they should be good.  v4l2 was 'wrong' to start with which is why I corrected it.