goodrobots/maverick #945: webrtc doesn't work on boot

fnoop, 2020-04-16:
On boot:
 - visiond service starts up first and then webrtc service
 - visiond starts rtsp resource, and connects to itself as a client to keep the rtsp stream open
 - visiond starts a websocket server for janus to connect to and dump events
 - visiond tries an initial http connection to janus api to see if it's available/present
 - webrtc starts up and connects to visiond rtsp stream
 - webrtc should fire an event to the visiond websocket server saying it's starting up.
 - visiond should take the websocket event, recognise that webrtc is up and running, and add a zeroconf broadcast to advertise the webrtc stream

However, this isn't happening.  Boot timing issues mean that visiond initial http query to janus api fails because janus isn't started yet, and webrtc initial start event seems to happen before visiond websocket server is active.

Need to fix this so visiond reliably picks up that webrtc is present and broadcasts zeroconf.

fnoop, 2020-04-16:
Changed visiond to Type=notify service type, and added sdnotify to notify the systemd service when the visiond pipeline has been fully initialised and is ready for connections.
It also notifies the systemd service status, eg:
```
Status of service: visiond
● maverick-visiond.service - Maverick Vision Daemon
   Loaded: loaded (/etc/systemd/system/maverick-visiond.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/maverick-visiond.service.d
           └─library-path.conf, openssl.conf, path.conf, typelib-path.conf
   Active: active (running) since Thu 2020-04-16 23:25:08 BST; 1min 35s ago
 Main PID: 2085 (python3)
   Status: "Automatic Pipeline Initialisation Complete"
```

fnoop, 2020-04-16:
This isn't working, webrtc is still running before visiond:
visiond: `Active: active (running) since Thu 2020-04-16 23:34:44 BST; 6min ago`
webrtc: `Active: active (running) since Thu 2020-04-16 23:34:40 BST; 7min ago`

fnoop, 2020-04-16:
Setting webrtc service to only start after visiond seems to have done the trick - it only starts directly after visiond has fully started (and the websocket server is therefore up and running and waiting for webrtc events).

closed 2020-04-16 by fnoop

SamuelDudley, 2020-04-17:
Confirming this is now working nicely on my pi3b+. Including streaming via webrtc and rtsp at the same time.

fnoop, 2020-04-17:
Awesome, thanks for the feedback :)