goodrobots/maverick #947: How to create jetson nano image?

fnoop, 2020-04-19:
Nano sd card has lots of odd small partitions that breaks the createimage.sh script:

```
[dev] [mav@maverick-ubuntuvm ~/software/maverick/manifests/maverick-modules/base/files]$ sudo ./createimage.sh sdb /var/tmp/maverick-nano-1.2b3.img.xz
Source Disk: sdb
Source Partitions:
NAME    MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sdb       8:16   1 29.3G  0 disk
├─sdb1    8:17   1 29.3G  0 part
├─sdb2    8:18   1  128K  0 part
├─sdb3    8:19   1  448K  0 part
├─sdb4    8:20   1  576K  0 part
├─sdb5    8:21   1   64K  0 part
├─sdb6    8:22   1  192K  0 part
├─sdb7    8:23   1  384K  0 part
├─sdb8    8:24   1   64K  0 part
├─sdb9    8:25   1  448K  0 part
├─sdb10   8:26   1  448K  0 part
├─sdb11   8:27   1  768K  0 part
├─sdb12   8:28   1   64K  0 part
├─sdb13   8:29   1   80K  0 part
└─sdb14   8:30   1  128K  0 part
Specify partition to shrink: 1
Partition data (sdb1): sdb1 sdb1 8:17 ext4 cf9d96ca-f7c4-45f5-9064-652345026106 0fc63daf-8483-4772-8e79-3d69d8477de4 APP 2e1b80b2-ce34-4a0b-98d8-5d801647887a 128 0 1 1 29.3G root disk brw-rw---- 0 512 0 512 512 1 mq-deadline 2 part 0 0B 0B 0 0B 1 sdb block:scsi:usb:pci none
Checking partition:
fsck from util-linux 2.31.1
e2fsck 1.44.1 (24-Mar-2018)
/dev/sdb1: clean, 655613/1564672 files, 5185279/6249123 blocks
Total blocks: 6249123, Used blocks: 5185279, Free blocks: 1063844
New size of resized filesystem = Used blocks (5185279) + 5% = 6222334

Resizing partition filesystem
resize2fs 1.44.1 (24-Mar-2018)
Resizing the filesystem on /dev/sdb1 to 6222334 (4k) blocks.
Begin pass 2 (max = 4118)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 191)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/sdb1 is now 6222334 (4k) blocks long.

Current size of partition: Start=28672
18432
20480
22528
24576
26624s, End=61413375
19327
22015
22655
24735
26879s
New size of resized partition = Start=28672
18432
20480
22528
24576
26624s, End=49778672s

Resizing partition
Warning: Shrinking a partition can cause data loss, are you sure you want to continue?
parted: invalid token: 18432
Yes/No? n
parted /dev/sdb unit s resizepart 1 28672
18432
20480
22528
24576
49805296 yes

Creating image file /var/tmp/maverick-nano-1.2b3.img.xz.xz from resized disk sdb
dd if=/dev/sdb bs=512 iflag=fullblock count="28672
18432
20480
22528
24576
49805296" | nice -n 10 xz -6 --verbose --threads=0 > /var/tmp/maverick-nano-1.2b3.img.xz.xz
dd: invalid number: ‘28672\n18432\n20480\n22528\n24576\n49805296’
  100 %                  32 B / 0 B > 9.999

Syncing kernel buffers

Done!
```

fnoop, 2020-04-19:
https://forums.developer.nvidia.com/t/shink-system-image-of-jetson-nano-for-smaller-size-sd-card/80522/16
https://forums.developer.nvidia.com/t/no-jetson-image-disk-creator-sh/118525/2

fnoop, 2020-04-19:
https://pythops.com/post/create-your-own-image-for-jetson-nano-board

fnoop, 2020-05-06:
Created image using nvidia script but how to expand filesystem on first boot?

fnoop, 2020-05-06:
bzip2 reduces end img from 20g to 8g, definitely worth doing for distribution

fnoop, 2020-05-06:
./rootfs/usr/lib/nvidia/resizefs/nvresizefs.sh

fnoop, 2020-05-06:
Also a script here:
https://forums.developer.nvidia.com/t/jetson-nano-sd-card-partitions-can-not-extend/73032/38?u=fnoop

fnoop, 2020-05-06:
```
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns: Collecting aiosqlite
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns:   Downloading https://files.pythonhosted.org/packages/bf/c5/25e3430a0f00b57a92d6ac485e5fb480a5ecbaa87069afcaf0a4d201d092/aiosqlite-0.13.0-py3-none-any.whl
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns: Installing collected packages: aiosqlite
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns:   Found existing installation: aiosqlite 0.10.0
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns:     Uninstalling aiosqlite-0.10.0:
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns:       Successfully uninstalled aiosqlite-0.10.0
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns: ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: 'top_level.txt'
Notice: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns: Consider using the `--user` option or check the permissions.
Error: 'pip3 --log /tmp/pip.log install --upgrade   --upgrade --disable-pip-version-check  aiosqlite' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_web::Maverick_api/Install_python_module[api-aiosqlite]/Python::Pip[api-aiosqlite]/Exec[pip_install_api-aiosqlite]/returns: change from 'notrun' to ['0'] failed: 'pip3 --log /tmp/pip.log install --upgrade   --upgrade --disable-pip-version-check  aiosqlite' returned 1 instead of one of [0]
Notice: /Stage[main]/Maverick_web::Maverick_api/File[/etc/systemd/system/maverick-api@.service]: Dependency Exec[pip_install_api-aiosqlite] has failures: true
```

fnoop, 2020-05-06:
```
mkdir nano_build
cd nano_build
wget -O https://developer.nvidia.com/embedded/r32-2-3_Release_v1.0/t210ref_release_aarch64/Tegra210_Linux_R32.2.3_aarch64.tbz2 |tar -xjspf -
rsync -avz <maverick.nano.sd.root>/*  Linux_for_Tegra/rootfs
```

fnoop, 2020-05-07:
Mount dd image as vmdk:
https://askubuntu.com/a/802491

fnoop, 2020-05-07:
Todo:
 - Rebase on Jetpack 4.4DP
 - Use external apt repos, get rid of local repos
 - Strip out a lot of crap to reduce the size, currently ~20gb.
 - Expand root fs on first boot
 - Add optimised components, in particular opencv with cuda support

fnoop, 2020-05-07:
https://elinux.org/Jetson_Zoo

fnoop, 2020-05-07:
Jetpack 4.4 still has local cuda and visionworks repos.

fnoop, 2020-05-07:
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fquick_start.html%23wwpID0EXHA
https://docs.nvidia.com/jetson/jetpack/install-jetpack/index.html
https://developer.nvidia.com/embedded/linux-tegra

fnoop, 2020-05-08:
OpenCV build recipe:
https://github.com/mdegans/nano_build_opencv

fnoop, 2020-05-10:
Rebased on Jetpack 4.4, moved to online repos away from local repos, removed various doc/dev packages.  bz2 image down to 6.8gb, uncompressed still over 16b so needs a 32gb card.  Probably as good as we'll get, as the nvidia stuff is so massive.

fnoop, 2020-05-12:
Rebase of jetpack 4.4 works fine until image created:
https://forums.developer.nvidia.com/t/jetpack-4-4-created-image-with-jetson-disk-image-creator-sh-will-not-boot-past-nvidia-logo/122815

fnoop, 2020-08-16:
Jetpack 4.4 rebase image recreation works fine now with latest driver pack.  Have to specify nano hardware revision, set to B01 (300).  Don't know if this means it won't work at all on previous versions.

Still need to expand FS on boot.

fnoop, 2020-08-18:
Added firstboot mechanism (~/software/maverick/bin/maverick-firstboot.sh) which for tegra platform calls the root partition/fs expansion and notifies user that it needs a reboot to activate.

fnoop, 2021-11-10:
Image creation script doesn't work with latest 32.6.1 (Jetpack 4.6) release.
https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3261/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/flashing.html#wwpID0E0PI0HA

Jai-GAY, 2022-04-07:
I am curious what does this **image** in this statement refers to.
Where I can download this image.

[Write the image files to the SD card (no need to uncompress if you use Etcher)](https://github.com/goodrobots/maverick/blob/stable/docs/current/README.md#jetson-nano-instructions)