goodrobots/maverick #973: No space left on device --Pi Zero

mooneyj, 2020-07-08:
First configure of Maverick 1.2.0 -raspberrylite on a Pi ZeroW struggled to configure. The swap space was full most of the time.
Do I need to explicitly disable everything I don't need in the localconf.json, such as SITL, ROS and the flight controller firmware modules?

```
Notice: Applied catalog in 39419.03 seconds

Maverick finished, happy flying :)

Traceback (most recent call last):
  File "/opt/ros/melodic/_setup_util.py", line 295, in <module>
    sys.stdout.flush()
IOError: [Errno 28] No space left on device
Failed to run '"/opt/ros/melodic/_setup_util.py" configure': return code 1
```

fnoop, 2020-07-08:
Hi, if you've got the raspberrylite download then no it should be configured for a pi zero, you don't need to disable anything.

What size SD card is it?  Has it expanded the main partition OK? (`df -h`)

mooneyj, 2020-07-09:
It hasn't expanded the main partition, it shows 16GB for a 64GB SD card.

`Filesystem      Size  Used Avail Use% Mounted on
/dev/root        16G   15G     0 100% /
devtmpfs        181M     0  181M   0% /dev
tmpfs           185M     0  185M   0% /dev/shm
tmpfs           185M   15M  171M   8% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           185M     0  185M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   54M  199M  22% /boot
tmpfs            37M     0   37M   0% /run/user/6789
`

I will do it manually using raspi-config: the expand_rootfs tool.
Thanks for having a look. 

I've attached the full output of $ maverick configure. 
[maverick_configure.log](https://github.com/goodrobots/maverick/files/4898525/maverick_configure.log)
The first line was:
`Warning: Scope(Class[Maverick_hardware::Raspberry]): Root filesystem/partition needs expanding, *please reboot* when configure is finished.
`
Perhaps it should prompt the user to stop the configure and resize using raspi-config: expand_rootfs. 

The error message is located in [/manifests/maverick-modules/maverick_hardware/manifests/raspberry.pp](https://github.com/goodrobots/maverick/blob/3025efaf593dbcbf69eb063a1bd7de641b8a3058/manifests/maverick-modules/maverick_hardware/manifests/raspberry.pp):

```
    if ($expand_root) {
        # $rootpart_expanded and $rootfs_expanded are facts provided by facts.d/filesystems.py
        if str2bool($::rootpart_expanded) == false or str2bool($::rootfs_expanded) == false {
            exec { "raspberry-expandroot":
                command     => "/usr/bin/raspi-config nonint do_expand_rootfs; echo 'done' > /etc/raspi-expandroot",
                unless      => "/bin/grep 'done' /etc/raspi-expandroot",
                require	    => Package["raspi-config"],
            }
            warning("Root filesystem/partition needs expanding, *please reboot* when configure is finished.")
```

mooneyj, 2020-07-09:
I've created a pull request to modify raspberry.pp 
[pull/974](https://github.com/goodrobots/maverick/pull/974)

mooneyj, 2020-07-10:
Is it possible to disable SITL wi

mooneyj, 2020-07-10:
accidental close sorry

fnoop, 2020-07-13:
So the fs expansion is a little complicated on the raspberry platform:
 - raspi-config is used to expand the filesystem:
   https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/maverick_hardware/manifests/raspberry.pp#L120
 - Maverick has 'facts' that are used as part of the puppet (configure) run to determine the current state of the system.  There are two custom facts that determine if the root partition, and the root filesystem have been expanded:
```
[flight] [mav@maverick-raspberry ~/software/maverick/manifests/maverick-modules/maverick_hardware/facts.d]$ ./filesystems.py
rootpart_expanded=True
rootpart_device=mmcblk0
rootpart_partition=mmcblk0p2
rootpart_partno=2
rootfs_expanded=True
```
If either of these are set to False, then the expansion (through raspi-config) is performed when configure is run, and the user advised to reboot to activate the expansion.

This is usually done in bootstrap environment when building from scratch, after which a reboot is done to activate the expansion before carrying on with a full flight or dev build.  For a download image, the initial `maverick self-update; maverick configure` should do the same thing - it should automatically take care of the expansion and then advise the user to reboot.
 - The expansion writes a marker file (/etc/raspi-expandroot) to indicate the expansion has been done so it can be ignored in the future by other tools.  

In your case it has failed because your SD card has run out of space *during* the initial configure, clearly we're riding the 16Gb line too finely.

I think what we could do here to improve the 'UX' is what other platforms do, which is to automatically call the expansion on first boot and reboot automatically (if necessary) without any intervention from the user.  There is already a firstboot script that we can use for this:
https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/base/manifests/maverick.pp#L170

fnoop, 2020-07-13:
Note: the firstboot script is run through a systemd boot manifest (maverick-firstboot) and enabled by default.  The manifest only runs if a filesystem marker is placed (`/boot/maverick`) - similar to how Raspbian uses `/boot/ssh` or `/boot/wpa_supplicant.conf` markers, but this file isn't placed by anything so never gets called.  We could place it through the preprelease script that we use to prepare all the releases (https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/base/files/preprelease.sh), and add the fs expansion to the firstboot actions instead of the current configure.  Perhaps forcing a lengthy configure automatically isn't a good UX.

fnoop, 2020-07-14:
Changed maverick-firstboot service to run a script instead of a `maverick configure`: ~/software/maverick/bin/maverick-firstboot.sh.   This script is then symlinked to a platform specific script according to the platform, ie. on raspberry it's linked to manifests/maverick-modules/maverick_hardware/files/raspberry-firstboot.sh.

On the raspberry, this then checks for a filemarker (/etc/raspi-expandroot) and if it doesn't exist, it expands the root partition, then the filesystem, then it automatically reboots.

The preprelease.sh script (which is used to create the dist images) removes the filemarker, so when users download the image and first boot it, it should automatically expand the partition/filesystem and then reboot.

fnoop, 2020-08-19:
SD card partition is now fully expanded on first boot and automatically reboots, so happens invisibly to the user.  Added to quick start and module (hardware) docs.

closed 2020-08-19 by fnoop