goodrobots/maverick #975: [feature] maverick start/stop/disable/enable all

julled, 2020-07-09:
Hi ! 

i am so glad you do this awesome image with all necessary stuff to make automated drones. Everytime i think i need a specific function, i later understood you already implemented it within this image.Thank you so much for this! 

I really would like to have a  generic "maverick start/stop/disable/enable all" function to be able to have a main switch for all the stuff. Do you think this could be implemented?

fnoop, 2020-07-10:
Hi, happy you're finding something useful here :)

There's already a `maverick start all` and `maverick stop all`:

```
[dev] [mav@maverick-nano ~]$ maverick stop all

Maverick - UAV Companion Computer System - Version 1.2.0-beta

WARNING: Maverick is using branch:master, not stable

Stopping all services
Stopping service maverick-api@system.service:  Stop: Success
Stopping service maverick-apsitl@dev.service:  Stop: Success
Stopping service maverick-cloud9.service:  Stop: Success
Stopping service maverick-collectd.service:  Stop: Success
Stopping service maverick-discovery.service:  Stop: Success
...
```
I can't remember when this was added - it might only be available in the 1.2beta (available here: https://goodrobots.github.io/maverick/next/#/)

There's not an equivalent for 'enable all', I'lll have a look and see if that's a good idea or not - we might have left that out originally because it's a bad idea to enable/disable all.  I can see the case for disable.  It might also have been around the different mavlink proxy services, which are now merged into a single service.

fnoop, 2020-07-14:
Updated `maverick` with better service control/checks, and also added a `maverick disable all`.  I don't think it's a good idea to have `maverick enable all` because there are so many available services that wouldn't be a good idea to start all together.

closed 2020-07-14 by fnoop

julled, 2020-07-14:
thx alot!