goodrobots/maverick #984: Configure failure on pi3b+

SamuelDudley, 2020-08-12:
Current master branch failed to configure for dev environment.
```
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: Traceback (most recent call last):
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/verbs/catkin_config/cli.py", line 149, in main
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     remove=opts.remove_args)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/context.py", line 247, in load
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     ctx.load_env()
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/context.py", line 442, in load_env
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     extended_env = get_resultspace_environment(self.extend_path, quiet=False)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/resultspace.py", line 96, in get_resultspace_environment
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     "file)." % result_space_path
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: OSError: Cannot load environment from resultspace "/opt/ros/current" because it does not appear to be a catkin-generated resultspace (missing .catkin marker file).
```
The above appears to be the start of it and the remaining steps fail due to this error. Full output attached.

[failed_configure.txt](https://github.com/goodrobots/maverick/files/5062574/failed_configure.txt)

SamuelDudley, 2020-08-12:
related to #977 ?

fnoop, 2020-08-15:
Could you do a self-update and configure and try again?  I can't replicate this.

SamuelDudley, 2020-08-15:
same result unfortunately...
```
dev] [mav@maverick-raspberry ~]$ maverick self-update

Maverick - UAV Companion Computer System - Version 1.2.0-beta4

WARNING: Maverick is using branch:master, not stable

Updating maverick: remote: Enumerating objects: 54, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 30 (delta 24), reused 21 (delta 15), pack-reused 0
Unpacking objects: 100% (30/30), done.
From https://github.com/goodrobots/maverick
   5a88ccb7..66acef7b  master     -> origin/master
   cd18863a..8fd5bbb2  stable     -> origin/stable
Updating 5a88ccb7..66acef7b
Fast-forward
 conf/sample-nodes/maverick-raspberrylite.json             |  2 +-
 manifests/maverick-modules/base/files/preprelease.sh      |  2 ++
 manifests/maverick-modules/maverick_ros/manifests/ros1.pp | 41 +++++++++++++++++------------------------
 3 files changed, 20 insertions(+), 25 deletions(-)
Updating visiond: Already up to date.
Updating maverick-web: Already up to date.
Updating maverick-web-dist: Already up to date.
Updating maverick-api: Already up to date.
Updating maverick-discovery: Already up to date.

Maverick update completed

[dev] [mav@maverick-raspberry ~]$ maverick configure

Maverick - UAV Companion Computer System - Version 1.2.0-beta4

WARNING: Maverick is using branch:master, not stable

Environment marker set and is being used to set maverick environment: dev
Warning: apt package cache has not been updated within the last day, running update
Maverick Environment:    dev 
Proceeding to update system configuration - please be patient, this can take a while..

Notice: Scope(Class[Maverick_intelligence::Pytorch]): Pytorch only supported on x86_64/amd64 platform for now.
Notice: Compiled catalog for maverick-raspberry.local in environment dev in 26.18 seconds
Notice: /Stage[main]/Maverick_security::Disable_services/Service[alsa-state]/ensure: ensure changed 'running' to 'stopped'
Notice: /Stage[main]/Maverick_ros::Ros1/File[/etc/profile.d/30-ros1env.sh]/ensure: defined content as '{md5}b34e63aabfbda15046736bffc781d72e'
Notice: /Stage[main]/Maverick_ros::Ros1/File[/etc/profile.d/30-ros-env.sh]/ensure: removed
Notice: /Stage[main]/Maverick_ros::Ros1/File[/etc/profile.d/31-ros-env.sh]/ensure: defined content as '{md5}12a9cab712ef5713b8524facc1ead116'
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: Traceback (most recent call last):
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/verbs/catkin_config/cli.py", line 149, in main
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     remove=opts.remove_args)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/context.py", line 247, in load
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     ctx.load_env()
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/context.py", line 442, in load_env
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     extended_env = get_resultspace_environment(self.extend_path, quiet=False)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/resultspace.py", line 96, in get_resultspace_environment
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     "file)." % result_space_path
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: OSError: Cannot load environment from resultspace "/opt/ros/current" because it does not appear to be a catkin-generated resultspace (missing .catkin marker file).
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: 
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: During handling of the above exception, another exception occurred:
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: 
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns: Traceback (most recent call last):
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/bin/catkin", line 8, in <module>
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     sys.exit(main())
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/commands/catkin.py", line 272, in main
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     catkin_main(sysargs)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/commands/catkin.py", line 267, in catkin_main
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:     sys.exit(args.main(args) or 0)
Notice: /Stage[main]/Maverick_ros::Ros1/Exec[ros-apriltag-catkin-init]/returns:   File "/srv/maverick/software/python/lib/python3.7/site-packages/catkin_tools/verbs/catkin_config/cli.py", line 167, in main
[... cut ...]
Notice: /Stage[main]/Maverick_analysis::Influx/Service[influxdb]/enable: enable changed 'true' to 'false'
Notice: /Stage[main]/Maverick_dev::Apsitl_apdev/Maverick_dev::Apsitl[apdev]/Maverick_ros::Rosmaster[apdev]/Service[maverick-rosmaster@apdev]/ensure: ensure changed 'stopped' to 'running'
Notice: Applied catalog in 246.75 seconds

Maverick finished, happy flying :)

-bash: /opt/ros/noetic/setup.bash: No such file or directory
[dev] [mav@maverick-raspberry ~]$ 
```
I'll re flash the latest image again and go from the start.

SamuelDudley, 2020-08-17:
Grabbed the latest beta. No longer an issue. Closing

closed 2020-08-17 by SamuelDudley