goodrobots/maverick #992: Flip/rotate video - visiond

collumnet, 2020-09-15:
Is it possible to flip or rotate the camera image in visiond with the "videoflip" gstreamer switch? Is there a param in ~/config/vision/visiond.conf, or can this be achieved pipeline_override?

fnoop, 2020-09-17:
Ideally this would be done in the source using hardware.  eg. raspberry can do rotation in hardware (raspistill/raspivid) but the v4l2src gst interface doesn't support it.  Possibly rpicamsrc does but will not be included until later version of gstreamer.  Similarly, nvarguscamerasrc for jetson csi (eg. raspberry camera on jetson nano) doesn't seem to support hardware rotation either.

So supporting videoflip gst module is a good idea, although it does the flip in software.  We should support hardware option where available - eg. on jetson platform we can use `nvvidconv` flip-method parameter.

fnoop, 2020-09-18:
Added 'rotate' config option in https://github.com/goodrobots/visiond/commit/13126e042b76088b2b4f4c1d6b0a602fe02491d8

Specifically uses nvvidconv.flip-method on jetson platform and forces post-rotate caps to allow connecting to encoder component, otherwise uses software videoflip.video-direction which still needs to be tested on other platforms (raspberry, primarily).

fnoop, 2020-09-18:
Confirmed working on raspberry

closed 2020-09-18 by fnoop