goodrobots/maverick #998: Can't find RPI image

alessandrogenoni, 2021-07-27:
Hi! I tried downloading maverick from the website (https://goodrobots.github.io/maverick/current/#/?id=installation) for a Raspberry Pi only to find that the link is broken and redirects to a generic page where

matthewoots, 2021-07-31:
I wanted to ask the same question too as @alessandrogenoni, is there any `image` available? Seems like **all** the links are broken

fnoop, 2021-08-07:
Hi, this project has taken a bit of a break and a few things have degraded.  It's back in development now, will post new downloads shortly.  Thanks :)

RotorBoy2020, 2021-08-25:
@fnoop Any update as to when the links will be live again? None of the downloads seem to be working at the moment

fnoop, 2021-08-29:
Hi, I've hit a few snags getting the software updated - there have been some major changes to the way some of the components are built and configured.  Actively working on it, will try and get a new release out ASAP.

RotorBoy2020, 2021-09-07:
@fnoop I know you guys are probably super busy working hard on the next version of Maverick OS, but any update on when the next version of Maverick will be available? Are you posting updates anywhere that I can follow? 

Thanks for all the great work you and your team are doing! Really appreciate it!

fnoop, 2021-09-07:
Hi @RotorBoy2020 - I've got stuck on two problems currently!  They're both fairly big problems to do with completely new build systems for two major components (gstreamer and ROS).  I'll do my best to get a release out by this weekend.

RotorBoy2020, 2021-09-08:
Thank you! We really appreciate your hard work!

fnoop, 2021-09-13:
Quick update: I've got a (desktop) VM build finally updated and running, and I'm now doing a full raspberry build.  This takes a day or so just for the silent build and will no doubt need some raspberry specific fixes.  But it's on it's way :)

mishaturnbull, 2021-09-14:
I have a 3B+ to test on if needed... it's sole purpose in life is Maverick, so let me know if I can help!

fnoop, 2021-09-20:
Apologies for delay, there have been quite a few updates/changes to some of the underlying components that took a while to work out.  I've uploaded a new image for Raspberry Pi (multi-cpu, ie not the zero or pi1/2):
http://mvrk.org/downloads/maverick-1.2.0beta5-raspberry.img.xz

It's not perfect by any means but any feedback gratefully received!

mishaturnbull, 2021-09-20:
[image]

First impression: it works!

I'll keep poking around and configuring (gotta get some hardware in before I can actually fly it... the voltage regulator I had isn't stable enough to run a Pi in flight) and let you know if I come across anything I can't resolve.

Thanks so much for all your work!  It does not go unappreciated!

fnoop, 2021-09-21:
@mishaturnbull great start :)

It doesn't look like you've got a camera hooked up - do you have one?  It would be great to get feedback on:
 - Loading the CA certificate from the front page fo the website, and
 - Getting the video to work

Both these were really horrible to implement technically, so it would be really good to get some feedback on how easy or not easy it is to use in person, with any suggestions of how to improve it!

mishaturnbull, 2021-09-21:
Yup, at the moment no camera... I have one on an Octopi I can pull off it... since that printer is now in a permanently-dark room, the camera doesn't do much good there :)

I did in fact load the CA cert shortly after this screencap.  It was *almost* perfectly in-line with the instructions given -- only one minor difference: when Firefox downloaded the image, the "import" button was greyed out.  I re-downloaded it, this time using the 'save file' option instead of 'open in Firefox', then went to my settings and imported it.  It worked fine that way.  I suspect, however, that this is not your fault -- I've made some very in-depth changes to my browser, and this is likely the outcome of one of those...

That said, here's what I've noticed with the certificate loaded:
* Home page works over SSL, albeit emitting a warning about domain mismatch.  I believe this is because the network I'm on doesn't appear to allow any of the autodiscovery type addresses -- `maverick-raspberry.home`, `.local`, `.lan`, etc all don't resolve.  This is *not* unique to Maverick; also happens with my Octopi, and therefore I just go to the IP address, which I think makes the cert not match.
* Grafana won't load over SSL whatsoever.  I can load & login over regular HTTP, though.

To be honest, the first thing I did next was dump a metric ton of dataflash logs into the inbox folder (about 2GB).  From there, I was hoping I'd be able to peek at something in Grafana, but it doesn't look like any of the dashboards have come up yet.  It's very possible the importer is still chewing through that data, though... any way I can check on that process?

Thanks again for all the work!

fnoop, 2021-09-26:
Hey @mishaturnbull - thanks for the feedback!
Grafana was a bit broken, both the provisioning and the log importing - it's changed a lot since the initial work tying the systems/data together.  I've updated a bunch of stuff (#1015) and it should work now, but you probably have to delete the existing databases in order for it to do so:
```
maverick self-update
rm ~/data/analysis/mavlog-meta.db
rm ~/data/analysis/grafana/grafana.db
maverick restart grafana
maverick restart mavlogd
maverick configure
```

I've raised #1018 for Grafana over SSL.

I'll update the CA loading page thanks (#1019), I'm sure the browsers have changed a bit since I first did it!  Great it mostly worked for you!

The autodiscovery is interesting - is your computer on the same network as the Pi?  Maverick sets up avahi to serve mdns requests - basically zeroconf broadcasts requests over multicast and various Avahi devices on the network listen to requests and answer if they recognise the request.  It would be unusual for a local router/hub/switch to filter those requests (albeit not impossible).  Can you try restarting avahi and then try again as soon after restarting avahi as possible?
```
sudo systemctl restart avahi-daemon
sudo grep avahi /var/log/daemon.log
```
What OS is your computer running?

mishaturnbull, 2021-09-28:
Glad I can help somehow!  :)

I've run through the update process as instructed, and am going to let it marinate on the logs overnight (I did a `touch` on all the files in the inbox folder to ensure it would trigger whatever needs triggering).  In the mean time, however:

* Grafana over SSL: Working!
* Autodiscovery: Working!  (`maverick-raspberry.local` only; `.home` and just `maverick-raspberry` didn't)

The autodiscovery is still a little bit my-network-related IMO, though, since also between tests I've installed and configured a network-wide Pihole server that's now handling all my DNS.  But... it works!  I'm running Linux on my laptop as my daily driver -- unfortunately (or fortunately) I don't have a Windows box I can test on.

Thanks again!

mishaturnbull, 2021-09-28:
It's had about 16 hours to chew on the data... unfortunately, I'm still not seeing any tables appearing in Grafana :(

I pulled the two databases you mentioned to purge, and linked them below if you want to look at it for analysis.  Not going to link directly to GitHub in case I accidentally put something in there that should be redacted... this way I can pull it after the fact :)

* [mavlog-meta.db](https://turnbull.link/drone/exports/mavlog-meta.db)
* [grafana.db](https://turnbull.link/drone/exports/grafana.db)

Taking a brief look at the mavlog meta, it looks like only one of the logs was processed successfully, although I don't know how to find more information on the errors.  The Grafana database was a little overwhelming, although it did appear to have some dashboards listed in the dashboards table.  (I did reboot as well to see if that had any effect - it did not)

Thanks!

fnoop, 2021-09-28:
I'm getting a 403 on the grafana.db :). Could you post ~/var/log/analysis/maverick-mavlogd.log, or at least have a look at it yourself for any obvious errors?
If you log into grafana as the mav user, you should have three dashboards 'starred' that should appear on the front page:
 - Flight Logs Index
 - Flight Errors Index
 - System Dashboard
If you click on the Flight Logs Index, you don't see anything here?  As it processes through your logs it should update this dashboard after each file, so you should see entries here within a minute or two of starting to process.
Do you see anything in the System Dashboard?  this should always show you cpu/load/memory data etc.

mishaturnbull, 2021-09-29:
*Oops*, my bad... didn't check the permissions.  Should be fixed now!

This is what I see logging into Grafana: 
[image]
The little 4-squares menu off to the left doesn't reveal much either :/

[maverick-mavlogd.log](https://turnbull.link/drone/exports/maverick-mavlogd.log) appears to have a few different errors - a couple InfluxDB timeouts, a parse error, and a KeyError all in the mix. (I checked the permissions on this one :) )

Thanks again!!

fnoop, 2021-09-29:
Huh that's odd - the databases look good except the flight dashboard is empty.  I can see the starred dashboards, maybe you need to restart grafana or reload/relogin to grafana to see them?
Can you try updating maverick, restarting mavlogd and grafana to make sure, and trying just a single log file?
```
maverick self-update
\rm ~/var/log/analysis/maverick-mavlogd.log
maverick restart mavlogd
maverick restart grafana
```
Then touch one of the logs and have a look again in the maverick-mavlogd.log.

mishaturnbull, 2021-09-29:
Unfortunately, same results.  New log file should be at [maverick-mavlogd-2.log](https://turnbull.link/drone/exports/maverick-mavlogd-2.log).  

This time, it looks like a KeyError in the importer process relating to [the particular log](https://turnbull.link/drone/exports/scrappy-20210624-cmot-try1.bin)... it opens fine for me in APMPlanner2.0 and in some other log analyzer that uses the DFReader class as well.

I have noticed a few other things while digging around, first off that mavlogd seems to be stopped when Grafana is started.  The order of commands to restart both is as follows:

```
rm ~/var/log/analysis/maverick-mavlogd.log
maverick status  # mavlogd & grafana are running
maverick restart mavlogd  # reports success on stop and start
maverick status  # both are running
maverick restart grafana  # reports success on both stop and start
maverick status  # grafana is running, mavlogd is stopped
maverick start mavlogd  # reports success
maverick status  # both are running again
```
It doesn't seem to affect anything other than running an extra command... but I'm wondering if there's a negative interaction somewhere.

There's also this message that appears in `/var/log/syslog`:

```
Sep 29 11:10:45 maverick-raspberry grafana-server[13885]: t=2021-09-29T11:10:45-0500 lvl=eror msg="failed to save dash
board" logger=provisioning.dashboard type=file name=dashboards error="Dashboard not found"
```

That seems to be the only message from Grafana in that log file, but I haven't yet managed to correlate it to a cause (i.e. does it get written to log when I go to the Grafana webpage or when I trip the event to start mavlogd reading the log?)

Again, thanks for all your help!  (I hope this comment works... Github won't allow me to preview it for some reason)

fnoop, 2021-09-29:
Hmm, the log is importing just fine - ignore the IndexError (I've raised #1021 to track that).  It immediately shows up in my flight log dashboard and I can see the data (although I also clearly need to update the import to recent Ardupilot fields - I haven't actually used recent firmware).
So why can't you see it?  Can you see the starred dashboards on the home page yet?  Have you found the Flight Logs dashboard?  According to your grafana database they're all there.  Are you definitely logged in as the mav user to grafana?  Have you tried logging out and back in again?
Feel free to drop by goodrobots/maverick in gitter if you want some realtime help :)

fnoop, 2021-09-30:
Since the grafana issue was mostly solved in gitter I'll close this issue in favour of more specific ones.
Thanks for the interest, and testing :)

closed 2021-09-30 by fnoop