goodrobots/maverick #1034: How to enable video streaming?

qproger, 2021-11-17:
Hello,
I flashed this OS.
Now how and where to enable video streaming? I want to test camera on my Raspberry 3B+.
Where are settings?
Where is documentation of how to setup video?

fnoop, 2021-11-17:
Hi, docs need updated a bit, but here:
https://goodrobots.github.io/maverick/current/#/modules/vision

There is now a web interface, try it by going to http://maverick-raspberry.local/, loading the SSL cert as directed on the front page, then going to the vision module.  It should just work out of the box on the raspberry, using raspberry camera.  If it doesn't, please do report any problems here and we'll try and work them out.

qproger, 2021-11-17:
In web interface on home page I don't see any "vision module". Only 3 buttons: config, system and video.

Explain please what to do / what to enable / where to go...

fnoop, 2021-11-17:
Go to video..

qproger, 2021-11-17:
Ok, created a dummy stream "wss://qqq:6796". In Stream Control enadled it.
See only black player with status "Stopped Bitrate0 K Packet Loss 0".

What now?

On Raspian testing camera is very simple, just enter "libcamera-hello" in console and watch...

fnoop, 2021-11-17:
> Ok, created a dummy stream "wss://qqq:6796". In Stream Control enadled it. See only black player with status "Stopped Bitrate0 K Packet Loss 0".
> 
> What now?

Have a look in the browser console, it could be a lot of things but that's the obvious place to start looking.

> On Raspian testing camera is very simple, just enter "libcamera-hello" in console and watch...

Sure, but not terribly useful on a UAV flying through the air at 50mph half a mile away.  Remote video is hard, remote video through a browser is even harder - last time I looked pretty much nobody else was offering this.  This project aims to make it click button easy but it's in early stages of testing.

qproger, 2021-11-17:
What I have:

```
Error: API qqq does not have a hostname set, please upgrade maverick-api
Firefox cannot establish connection with server wss://maverick-raspberry.local:6002/.
Firefox cannot establish connection with server wss://qqq:6796/.
```