goodrobots/maverick #1035: Download for images is broken

Vabe7, 2022-02-23:
On the documentation website the downloadlink for the images do no longer work:
https://goodrobots.github.io/maverick/current/#/?id=os-images

arct1cOne, 2022-04-04:
Have the same issue.
Needed Raspberry Pi lite image

CCSnell, 2022-06-26:
The reason is because of broken SSL configuration for the download site. Not sure what it is, but if you wanna download anyway. Right Click on the link, and "Save Link As". Some browsers will provide a warning, but since the hash is provided next to the download you can confirm manually after download.

Phillweston, 2022-08-18:
I tried to download manually, but just a small file that cannot be used.
Here is some of the information shown by the network page.
This domain was recently registered at [Namecheap.com](https://www.namecheap.com/). Please check back later!
I wonder if the download link has expired?

mohammada315, 2022-08-28:
This issue has not been resolved yet

VEEROBOT, 2022-12-13:
Not able to download still. Seems like the link is removed. Tried right click-> save as, but no luck.

TonyStark2007, 2023-07-25:
Anyone has the link or a old copy of the image for rpi?

Seems no help whatsoever, plz share image link if you have one. Thank you.

rakonjac-uros, 2023-09-11:
Any updates? Can any images be found anywhere? @fnoop

packet43, 2024-03-20:
Does anyone have a link to the Raspberry PI version. I would love to help get this file hosted and provide some help. I know that certain people are using this and need it to be avalible. I'd really not like to have to bootstrap this myself if possible. Anyone?